supermerill/SuperSlicer #468: Print time estimate showing negative seconds amount

toskium, 2020-09-14:
### Version
2.2.53

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Klipper gcode flavor
DIY Printer

### Behavior
After slicing objects an estimated print time is usually displayed, but instead a negative amount of seconds is displayed.

[image]

#### Project File (.3MF) where problem occurs
[printtime_3mf.zip](https://github.com/supermerill/SuperSlicer/files/5216359/printtime_3mf.zip)

neophyl, 2020-09-14:
#364  #437

toskium, 2020-09-14:
Ah, sorry for the duplicates, I'll close it.

closed 2020-09-14 by toskium