supermerill/SuperSlicer #469: Temperature tower is missing degrees label at certain temps

toskium, 2020-09-14:
### Version
2.2.53

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Klipper
DIY Printer

### Behavior
When generating a temperature calibration tower with the following:

0 - up
7 - down
10 - steps

The temp tower starts at 225 and shows the degrees: 225, 215, 205, 195, 185 but the last three segments are missing the degrees label.

A similar problem exists when doing a temp tower with:

0 - down
7 - up
10 - steps

It starts at 295 on the lowest segment, but the label is missing. The first label shown is the second segment 285.

#### Project File (.3MF) where problem occurs
[temptower.zip](https://github.com/supermerill/SuperSlicer/files/5216393/temptower.zip)

supermerill, 2020-09-14:
yeah, i didn't create labels for evey possible temperature.
Below 185 and over 285 were considered by mysefl out-of range of useful temp.

Note that it doesn't means that the temperature isn't set, it just means that the label isn't printed so you have to substract/add it yourself. I didn't wrote something about hat in the note section?

Did you really need temeprature that low / high?

toskium, 2020-09-14:
Ah, thank you for the explanation. It wasn't clear to me you had to create the labels first.
I don't need it, I just found out by accident and thought it might be a bug so I opened the issue :-)
Can be closed from my point of view.

closed 2020-09-14 by supermerill