supermerill/SuperSlicer #471: Bridge between infill payers and speed modifiers

khumarahn, 2020-09-14:
Hi, I am trying to slice a benchy 2.2.53, coming from prusaslicer and looking for reduced speed for overhangs.

I see two immediate problems, both apparent on the screenshot (this is everything with default settings for a random printer, in this case voron v2). First is, there are bridge infill layers between the regular infill layers. This does not appear in prusaslicer.
[image]

Second is, a box modifier where only speed is modified creates additional perimeters around it. I was hoping to slow down the print in a particular area of overhangs by a modifier, but this changes the patterns quite a bit. This second problem does appear in prusaslicer, including the current git version.

neophyl, 2020-09-14:
Thats one of the problems with the modifiers.  The way they work is to split up the model so you get those annoying extra perimeters.

The pattern of support infill iirc is an option for putting a denser layer in before top fill.  If you look at the back of the Benchy where the flag pole is there is top fill there so its inserted the support for top layers.  If you go to Print Settings>Infill>Reducing printing time the options are there.  Supporting Dense Layer, change it from automatic to one of the others.  I think Anchored is the one you want.

supermerill, 2020-09-14:
> First is, there are bridge infill layers between the regular infill layers. This does not appear in prusaslicer.

It's the 'denser infill', you can uncheck it in "infill". It's to support solid surface when there is a low infill%.

> Second is, a box modifier where only speed is modified creates additional perimeters around it.

Currently, you can't change the speed in-the middle of an extrusion. So spitting surfacess the only way. It's a big task to allow it, and not that useful, so it may stay like that for a long time.

khumarahn, 2020-09-14:
@neophyl @supermerill Thanks for the explanation! Frankly, I thought that something is broken, but now things look brighter. I will try to use the settings of superslicer to make overhangs work without degrading layer adhesion.

closed 2020-09-14 by khumarahn