supermerill/SuperSlicer #473: [FR] Raft Settings

afkfurby, 2020-09-14:
Separate settings for:
- above raft speed (the first layer that is printed on the raft top layer)
- raft model z-distance (or separation distance) currently this is fixed at 0.3mm and can only be adjusted trought support z distance but the raft top need to have a different speed and distance to get a good first layer.

supermerill, 2020-09-14:
You can create a range modifier and change the bridge speed for the layer.
[image]

For a different z-distance for raft vs support, not possible, and I don't see why (and i don't want to touch to support code ><).

afkfurby, 2020-09-15:
I know that I can use range modifiers but that way I can not fix it for a specific profile (we are a company with 50+ printers and I try to kill S3D with SS) can you then add a variable for the fixed 0.3mm of the support z distance? that way I can change it :)

afkfurby, 2020-09-15:
It's in src/libslic3r/Slicing.cpp from line 149 to 179, there is also a FIXME, it uses 0.75* avarage nozzle size of all extruders printing on the interface layer of the raft.

if you could make an ifelse (if variable raft z distance is set use it otherwise use those 0.75*nozzle.

supermerill, 2020-09-15:
> I know that I can use range modifiers but that way I can not fix it for a specific profile (we are a company with 50+ printers and I try to kill S3D with SS) can you then add a variable for the fixed 0.3mm of the support z distance? that way I can change it :)

It's in my roadmap to add "modifiers scripts" into a profile for things like that. 
something like 
```
for all objects
modifier layers 0-0.2
    bridge_speed = 10
```
That will be applied on the platter (unless redefined).
#412

> the raft top need to have a different speed and distance to get a good first layer.
> can you then add a variable for the fixed 0.3mm of the support z distance?

Can you explain me why? I never use a raft, so i'm kind of a newb with rafts.

> It's in src/libslic3r/Slicing.cpp from line 149 to 179, there is also a FIXME, it uses 0.75* avarage nozzle size of all extruders printing on the interface layer of the raft.

I'm not sure this does what you think. and the 0.75 is fro the layer height of the raft. nothing to do with the interface.

afkfurby, 2020-09-16:
> It's in src/libslic3r/Slicing.cpp from line 149 to 179, there is also a FIXME, it uses 0.75* avarage nozzle size of all extruders printing on the interface layer of the raft. 
I'm not sure this does what you think. and the 0.75 is for the layer height of the raft. nothing to do with the interface.

Sorry, It's from line 141 to 147 to be correct. it checks if the support extruder does not use soluble material and uses a result of full z step height und nozzle diameter for "params.gap_raft_object".

> Different from the support-object interface distance, right?
Can you explain me why? I never use a raft, so i'm kind of a newb with rafts.

Because the whole object lies on the interface and is not supported anywhere else like "most" support, things like islands printed on support would better be handled as on top of a raft because of that. The first layer need to bond enought to the raft interface to stay in place but to not adhere to much so it could be peeled off easily.
With support_z_distance set to zero and printing ABS, my first layer prints nice, stays in shape und can be peeled off easily (resulting z_distance of 0.3mm). with a support_z_distance of 0.2mm and a resulting raft_z_distance of 0.5mm the first layer shrinks too much and gets shaped like \/. generally the first layer over a raft is printed slow, adjusted flow and (depending on material) medium to much cooling. 
And depending on how big the surface of the first layer ist, the z_distance needs to be adjusted in 0.01mm steps (I know most printers do not really can drive the z axis in that small steps) so the adhesion is not too big and you need to cut off the raft

supermerill, 2020-09-16:
> Because the whole object lies on the interface and is not supported anywhere else like "most" support

but there is also support that support their own islands.
Hence, it's better to use "support_z_distance set to zero " if it works and peels easily.
Does this setting value creates problems with supports?
I don't see why the offset could work for raft and not for supports.

supermerill, 2022-03-23:
raft model z-distance is now implemented in 2.4

closed 2022-03-23 by supermerill