supermerill/SuperSlicer #474: Poor seams on cylinder compared to PrusaSlicer

psonoda, 2020-09-14:
### Version
2.2.53
PIN-HOLE

### Operating system type + version
Win10 Pro fully patched

### 3D printer brand / version + firmware version (if known)
MK3S /  3.9.0

### Behavior
Poor seams on 24mm cylinder with 1.5mm walls

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
ss_Test_10mm*.3mf - See attached

[Poor_Seam.zip](https://github.com/supermerill/SuperSlicer/files/5220751/Poor_Seam.zip)

See attached images and .3mf files
131013.jpg seam from PrusaSlicer 2.2.0
131446.jpg seam from SuperSlicer

ss_test_10mm*.3mf is saved 3mf from SuperSlicer
test_10mm*.3mf is saved 3mf from PrusaSlicer

The ss_test file was loaded directly from the PrusaSlicer file. I have tried many of the seam setting and I could never get the seam looking near as good as PrusaSlicer. It's my first time using SuperSlicer and I really appreciate the awesome setting within SuperSlicer.

supermerill, 2020-09-14:
I'll try to print them and reproduce. Photos are blurry, I can't be sure of what i see.
The problem is that the seam is not strait?

psonoda, 2020-09-14:
Just tried it 2.2.51 and the seam is perfect. Yeah the seam is super wide and ugly. Looks underextruded to me

psonoda, 2020-09-15:
Okay, I see the difference. The issue happens between 52.1 and 53.0. It looks like on 52.1 (and PrusaSlicer) it's treating my inner layers as 2 perimeters and on 53.0 and 53.1 it's treating it as a single gap.
[image]
[image]

supermerill, 2020-09-15:
Use overlapping perimeters -> for all

psonoda, 2020-09-15:
Perfect, thank you!

closed 2020-09-15 by psonoda