supermerill/SuperSlicer #475: Feature request: reduce temperature for layers under a certain printing time

dexer12, 2020-09-15:
There is an option to slow down printing speed if the layer takes less then some seconds.
But often this is not enough in my opinion. So i would be great to get an option to reduce to printing temperature.

For example, printing a pyramid, you need the fast printing for the base and it will slow down speed more and more until the tip of the pyramid. But on the very top there are still 5 seconds or something like that for the last few layers. So it would be great to get an option to reduce the temperature to get a nice finished print here.

supermerill, 2020-09-15:
The speed of the temperature change... the effect on your material... too many problems with this. And i'm not sure this will be enough.

Did you tried it by modifying the gcode manually (just add a M104 command) ?

My solution: Add a tall thin cylinder and a other bigger one on top, near your object if you really need it.

dexer12, 2020-09-16:
Im printing two pyramids now, with your hint to add the M140 command, so they are both exact the same gcode file with just this change.
I'll post some photos after both single printed are finished.

dexer12, 2020-09-18:
I printed the pyramid 3 times, the left on is the last one, where i changed the temperature for the tip from about 230 degree to 190 degree. i think i need to go lower to around 180 degree the get better results.

[image]

I still think it would be great to have a modifier or something like that for such small layers.

supermerill, 2020-09-18:
you can add a custom gcode at the height you want.
in the preview view, go to your desired height, right-clic just at the right of the scroll bar. select custom code and put M104 S180

After re-slicing, you can see the change with the "temperature" preview color

aceycv8, 2020-09-19:
Hi Supermerill 

i'm using 2.2.53 and can't seem to get this option righ clicking on the scroll bar as i'd like this function too -is there something i should enable? i've got the application in expert mode

dexer12, 2020-09-19:
Okay, i think thats enough for me. I didnt know that before.
@aceycv8 you have to right click at the PLUS icon there you will get a small menu for color change, pause or custom gcode.

aceycv8, 2020-09-19:
thanks @dexer12  i see it thank you