supermerill/SuperSlicer #477: Seam-Sphere too big

dartrax, 2020-09-16:
### Version
2.2.53

### Operating system type + version
Win 10 x64

### Behavior
Thanks for making a up to date Slicer version with individual control over seam placement available! I'll use this from now on over my old Prusa 1.4 mod.

I've noticed the seam sphere being rather huge, it would be easier to guess the actual seam position if it would be smaller. The cause may be the large print bed (430 x 390), may be using the object size as size reference?

[image]

supermerill, 2020-09-16:
For the time being, you can change its size with the size widget at the bottom right.

dartrax, 2020-09-16:
Ah I didn't see that. Great!

closed 2020-09-16 by dartrax