supermerill/SuperSlicer #479: SS Crashes when slicing starts

afkfurby, 2020-09-17:
### Version
2.2.53.1

### Operating system type + version
windows 10

### Behavior
try to slice the attached project, SS crashes.
The loaded STL file does not matter.
I had this problem when switching from 2.2.52 to 2.2.53.1 with another profile and recreated all settings from ground up with the wizard and the problem dissapeared. Doesn't work this time.

Works with 2.2.52.
Cannot find any log files. Where are they stored or activated?

#### Project File (.3MF) where problem occurs
[Flow Test v3.zip](https://github.com/supermerill/SuperSlicer/files/5236940/Flow.Test.v3.zip)

neophyl, 2020-09-17:
I can confirm that the project as attached crashes that version of Slicer (on Win10 64 bit)  too.

What I have discovered is that if you turn off the External perimeter first option it then slices.  Turning that option on with the Apply on: both UN ticked allows it to slice.  Enabling either/Or of the outside only or inside only causes it to crash again.  

Why its doing so I will leave to others :)

afkfurby, 2020-09-18:
Is there a way to enable logging and/or that SS displays what it is doing while the progressbar is displayed?

I have it many times that it hangs itself while slicing but can not dig it down to something because of missing info.

supermerill, 2020-09-18:
Not without compiling the source code and using the debugger.
I tried to display more significant message than "slicing", but it's a bit complicated.

afkfurby, 2020-09-18:
funny, when using the command line version it tells you what it is doing :) at least a bit

closed 2020-10-22 by supermerill