supermerill/SuperSlicer #481: Feature Request/Question: Custom slicer variables

TD5023, 2020-09-18:
### Version
2.2.53

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
MakerGear M2 Rev E / Custom CoreXY

_Is this a new feature request?_
maybe

Would it be possible to add a section in each tab with "blank" variables that could be assigned as needed? The reason I'm looking for this feature is that I adjust z-offsets per filament material in the custom g-code section with M206. Currently, each filament has the number hard programmed in (except for PLA, which is my machine default). Whenever I calibrate, I have to go in and change each value. This isn't a big deal, but it would be nice if I could define it in the printer tab and then have the filament call that variable and perform the appropriate math. 

However, I'm going to be building a new CoreXY printer soon, and it will have a different offset. The filament adjustments will be the same, so having them call a per-printer value for the default offset instead of hard-coding them would be great.

Here is how I currently have the code:
[image]

Here is how I would like it to look:
[image]
Where print_var1 is the default M206 for each printer as assigned by me in the Printer Settings tab.

It could just be a page of empty boxes with standard names (print_var1, print_var2, fil_var1, etc.) and a note box next to them so the user can describe it. I don't know programming well enough to dig into the source code and force my own variables, but it would be really nice to have a little extra flexibility with custom gcode. I haven't found an accessible way of doing this now, but if one exists, I'd greatly appreciate some guidance.

supermerill, 2020-09-18:
You can't use z_offset directly and print M206 Z-0.06?
it's in printer ->general in expert mode

TD5023, 2020-09-19:
No, that doesn't work for my printer because my M206 default value is not zero. The M2 has a fixed z-endstop in the maximum y position with no adjustment screw, and the only way to make calibration moves is with M206 changes. Currently, its value is 11.30. If I were to zero out the offset on the printer, it would require starting z at -11.30 in the code, which the firmware will not allow. So, what I'm looking for is a way to add the 11.30 as a variable in the printers section that does not affect the body gcode, but can be used to perform math operations in filament custom gcode.

I hope I'm making sense. Also, for what it's worth, I did some gcode tinkering awhile back and I remember wishing I could define some variables for what I was trying to do (not offset related), but I can't remember what that was now. I could undoubtedly find several uses for this feature, but I could very well be alone in that.

ManuelMcLure, 2020-09-24:
@TD5023 The workaround I use for different filament offsets is to use `G92` to reset my Z=0. So in my PETG start gcode I do something like:
```
G0 Z2
G92 Z1.95
```
which has the effect of moving my Z offset up 0.05mm so PETG doesn't stick as hard to the bed.

TD5023, 2020-09-24:
@ManuelMcLure I like that. For some reason, it never dawned on me to use G92 for the movement axes. I've always just thought of them in absolute terms from the home position and only used G92 for filament resets. I'll do some adjusting later.

This should fix the example I gave, but I'd still like to eventually see the ability to define variables. However, what was already low-priority in terms of updates just got even lower.

TD5023, 2020-09-30:
@ManuelMcLure So, I just tried the G92 method and it turns out that it will not work with my setup. When a G92 is used to reset an axis, it automatically adds in any M206 value stored. So, if I run G0 Z180, then G92 Z180, and then G0 Z180 again, it will move up the by the value of the stored M206. The only way to make it stay put is to use G92 Z{180-11.3}, which goes back to my original issue of needing my machine offset value stored as a slicer variable. I was really hoping that would work, too :/

ManuelMcLure, 2020-10-02:
I see. I don't use `M206` at all, I just have my Z offset set correctly with `M851` and use `G92` to adjust Z in my startup code if I'm going to be printing PETG. But that works because I am homing with a probe.
If you're using Marlin and homing Z with an end stop switch you might be able to use `Z_MIN_POS` in the configuration file to take the place of needing to set `M206` for each printer in the slicer.

TD5023, 2020-10-04:
I just saw a typo in the original comment that might have led to some confusion. The M2 homes at the maximum z (Z_HOME_DIR 1), so it needs an M206 to know the buffer zone between the firmware maximum height and the actual endstop location. I could adjust the maximum print height to reflect the exact distance to Z0, but then I would have to change the firmware for every minor calibration move instead of simply resetting the M206 and committing it to memory via OctoPrint. As it is, I'd rather go through each filament in SS and update numbers individually than deal with altering firmware for normal calibration. 

I should note, too, that I've found other uses for my original request than just z offset, but it's all just for personal convenience and not necessary for adequate slicing.

supermerill, 2021-11-07:
seems to work
[image]

TD5023, 2021-11-07:
Digging back quite a bit, I see. To be honest, I had completely forgotten about this one. That's exactly what I was hoping for, though. Thanks!

supermerill, 2022-01-18:
I added the `default(var, value)` to avoid bugs when it's not defined.
after this one, you can use `exists(var)` that returns a boolean value, so you know if it's defined or the default value, so you can avoid creating a gcode line if the variable isn't defined.
All these have to be inside brackets, like all the macro stuff.

closed 2022-02-13 by supermerill

adawalli, 2022-05-08:
@supermerill  Is there an easy way to set a variable in the printer GCODE section? I know we have the default function...but what about something like

`{z_offset[0]=-.03}`

supermerill, 2022-05-18:
you can't set the variable because any change will be erased just after when the new region is loaded.
I have to create a kind of system of "override" that didn't exist yet.