supermerill/SuperSlicer #482: Ironing every infill layer (feature request)

LazaroFilm, 2020-09-18:
I'd love to see an option to iron every layer of a 100% infill. That would allow transparent filaments to become way more see through as seen in this post: https://community.ultimaker.com/topic/26593-please-include-an-iron-every-layer-option-for-clearer-prints/
Thanks!

supermerill, 2020-09-19:
duplicate of #38

in expert mode, solid infill -> ironing

LazaroFilm, 2020-09-20:
Ooh! I hadn't realized. Sorry for the duplicate, I swear I spent some time fidgetting with the settings first but I must have missed this. I have natural PETG on the way, I'll report back when i try this!

closed 2020-09-20 by LazaroFilm

amagr0, 2020-12-04:
Hello @supermerill ,
I'm doing a research project with metallic filaments, and one off the problems with the printed parts, is the air gaps between lines and layers, has you can see on this image.

[image]

I was looking for a solution, and it seams that the ironing feature could partialy resolve this. 
I already tried Cura and PS but without success, and then I found this post and I'm almost achiving what I indend.

The printed parts must be solids, and the toolpath should be as smooth as possible in a concentric way, with minimum of retactions. 

I have 3 possibilities, but none of them are giving me the intent results.
1. Only perimeters (concentric) - ironing not working
2. Only 100% Infill (concentric) - ironing not working
3. Only Top or Bottom Solid Layers - ironing working, but I cannot change pattern

[image]

How can I print concentricly and with ironing on every single layer?

Thanks and congrats for the amazing software you build.

supermerill, 2020-12-06:
It seems you are talking about the ironing post-process. This one will over-extrude, so i'm not sure it's what you want.

It's easier if you can build the source code, as you can make more exotic experiments.
For exemple, using the ironing infill pattern, you can change the pattern used for the first & second pass by changing the enum name in fillsmooth.hpp.
you can also add "hidden" smooth infill by copy-pasting the line 726 in printconfig.cpp two times.