supermerill/SuperSlicer #483: Print from height?

aceycv8, 2020-09-19:
Hi, 

i'm pretty new to the Prusa/Slic3r /Superslicer scene,

i'm looking how do i print from a set height?

e.g. if my print fails i want to resume it from it's height

supermerill, 2020-09-19:
You have to edit your gcode file manually.

copy your failed gcode print.
find the G0 ZXXXX
remove everything before it, but the first ~20 lines, before any G0/G1
print that.

note that you may encounter problems for the z levelling at startup.

supermerill, 2020-09-19:
Just a tough, you may be able to re-create the gcode with the slicer...

you can check the slicing -> allow empty layers
then you cut your object
then you move it on the z axis
but to get the right height, you have to do some math & experimentations.

aceycv8, 2020-09-19:
Thank you supermerill - that will work and i can work around that.
in s3d there is an option to start a print from height in mm and would work quite well but this works too

closed 2020-09-19 by aceycv8