supermerill/SuperSlicer #484: [FR] Add temperature and fan offset in the printer settings

iBLtz, 2020-09-19:
### Version
SuperSlicer_2.2.53_2020-09-18

### Operating system type + version
MacOS 10.15.6

### 3D printer brand / version + firmware version (if known)
Voron 2.4 Direct Drive/klipper 0.8.0

### Feature Request
I have a Voron and a lot of filament profile. I also often change the nozzle size to increase the print speed.
I usually print ABS at 245° with a 0.4 nozzle.
When printing ABS with a 0.8, in order to properly melt the ABS (25mm3/s), I need to increase the temperature (+10/15°, depends on the color) and to increase the fan as well. 

For every nozzle size, I need to have several copy of the same filament profile, where only the temperature and the fan is changing. 

Idee would be to have those offset values in the "printer settings" under "extruder", so I only need one profile pro filament. For every nozzle, I would just need one "profile printer" with a different nozzle size/temperature offset/ fan offset.

thx in advance :)

afkfurby, 2020-09-21:
would also appreciate that. Discovered this last week when i needed to change the fan speed to 100% at certain height to stop overheating of a very flat overhang and was very confused that this was not possible :)

supermerill, 2020-09-21:
@afkfurby You talk about issue #478, right? not this one?

afkfurby, 2020-09-22:
@supermerill oh fuck, yes #478 is the right one

yschroeder, 2020-09-23:
I think the main problem is (and this is true for many settings in FDM 3D printing) that all settings look like they are independent from each other while in reality they are not. These are however two example I also ran into recently. We have two printers, one with a V6 and 0.4mm nozzle, one with a Volcano and 0.8mm nozzle.

Both printers are able to handle vastly different amounts of material per second (10mm³/s vs. 30mm³/s). Luckily, I don't have to increase the temperature as the volcano is able to handle it. However, the excessive heat produced by triple the amount of material per second needs much more cooling.

supermerill, 2020-09-23:
> This list of settings goes on and on. In my case it would be better to have the maximum volumentric flow configured per printer...

moving the max_volumetric_speed from filament to extruder and adding a max_volumetric_flow_percentage into filament (so you can set 50% for abs) ?

yschroeder, 2020-09-23:
That might work!

iBLtz, 2020-09-23:
> > This list of settings goes on and on. In my case it would be better to have the maximum volumentric flow configured per printer...
> 
> moving the max_volumetric_speed from filament to extruder and adding a max_volumetric_flow_percentage into filament (so you can set 50% for abs) ?

that would be a nice change. I had always wondered why it was in the filament in the first place. 

I know it is not the right place to discuss it but since we are talking about Max Volumetric Speed, I have always wondered what kind of fantasy values is that 904mm3/s? 
[image]

> I think the main problem is (and this is true for many settings in FDM 3D printing) that all settings look like they are independent from each other while in reality they are not. These are however two examples I also ran into recently. We have two printers, one with a V6 and 0.4mm nozzle, one with a Volcano and 0.8mm nozzle. They can print off the same spool of filament so they should be able to use the same filament profile. But:
> 
> Both printers are able to handle vastly different amounts of material per second (10mm³/s vs. 30mm³/s). Luckily, I don't have to increase the temperature as the Volcano is able to handle it. However, the excessive heat produced by triple the amount of material per second needs much more cooling.
> 
> This list of settings goes on and on. In my case it would be better to have the maximum volumentric flow configured per printer...

I'm doing 0.8 with a dragon high flow (28mm3/s). the melting chamber is smaller than the volcano, that might explain why you do not need the extra temperature offset, because the filament has enough time to getting hot and melted. 
So I guess the temperature offset depends on the Hotend you are using. 
In your case it would be 0°, but I would still need the extra 10° bump :P

Regarding the cooling, well, it is independent from the hotend and we both have the same need.

yschroeder, 2020-09-23:
That hotend temperature offset would then also be in the extruder settings, right? Because as you said, your filament exists the nozzle too cold due to the high extrusion rate. The temperature configured for the material is correct, your printer just does not reach it.

To continue being off topic: I never understood what the sentences in the screenshot wanted to tell me anyway. Can some enlighten both of us?

iBLtz, 2020-09-23:
I would have placed the offset in this menu as well (maybe under the nozzle parameter since they are kind of related?)

yschroeder, 2020-09-23:
@supermerill already implemented it, see the commit above!

supermerill, 2020-09-23:
>That hotend temperature offset would then also be in the extruder settings, right?

yes, see op and commit:

> Idea would be to have those offset values in the "printer settings" under "extruder"

https://github.com/supermerill/SuperSlicer/commit/474ec2ff55bec05cc2637fac3216c9457526a49b#diff-7ba1c4d8a65c6810bc616aeee8ae5eb4

>  I never understood what the sentences in the screenshot wanted to tell me anyway. Can some enlighten both of us?

Never looked at it. The content seems wrong.

closed 2020-10-22 by supermerill