supermerill/SuperSlicer #486: Running SuperSlicer on Raspberry Pi

Pavulon87, 2020-09-19:
### Version
2.2.53.1-1-g95318d66d

### Operating system type + version
Linux raspberrypi 5.4.64-v7l+ #1340 SMP Fri Sep 11 14:15:31 BST 2020 armv7l GNU/Linux

### Behavior
Has anyone tried to run SuperSlicer on a Raspberry Pi?
After a lot of dependency problems I was able to complete the compilation.
I have tried:
`cmake .. -DCMAKE_BUILD_TYPE=Release -DSLIC3R_BUILD_TESTS=OFF -DSLIC3R_GTK=3`
or
`cmake ..  -DSLIC3R_STATIC=0   -DCMAKE_BUILD_TYPE=Debug   -DSLIC3R_FHS=1   -DSLIC3R_GTK=2   -DSLIC3R_PCH=0   -DCMAKE_INSTALL_PREFIX=/usr   -DCMAKE_INSTALL_LIBDIR=lib`
and after that
`make -j4
sudo make install`
The weird thing is that the GUI version is working, but command line not.
[image]
It is possible to test --help
`superslicer --help
SuperSlicer_2.2.53+UNKNOWN based on Slic3r (with GUI support)
https://github.com/prusa3d/PrusaSlicer`
But every time I try to use **--export-gcode** or **--slice** I get a "Segmentation fault"

Last lines of `strace "/usr/bin/superslicer" "/home/pi/.octoprint/uploads/cube.stl" -g`:

> openat(AT_FDCWD, "/home/pi/.octoprint/uploads/cube.stl", O_RDONLY) = 3
> fstat64(3, {st_mode=S_IFREG|0600, st_size=684, ...}) = 0
> fstat64(3, {st_mode=S_IFREG|0600, st_size=684, ...}) = 0
> _llseek(3, 0, [0], SEEK_SET)            = 0
> read(3, "Exported by Simplify3D 3.0.0 -  "..., 684) = 684
> _llseek(3, 684, [684], SEEK_SET)        = 0
> _llseek(3, 684, [684], SEEK_SET)        = 0
> _llseek(3, 684, [684], SEEK_SET)        = 0
> close(3)                                = 0
> gettimeofday({tv_sec=1600536483, tv_usec=617036}, NULL) = 0
> openat(AT_FDCWD, "/etc/localtime", O_RDONLY|O_CLOEXEC) = 3
> fstat64(3, {st_mode=S_IFREG|0644, st_size=2705, ...}) = 0
> fstat64(3, {st_mode=S_IFREG|0644, st_size=2705, ...}) = 0
> read(3, "TZif2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\v\0\0\0\v\0\0\0\0"..., 4096) = 2705
> _llseek(3, -1707, [998], SEEK_CUR)      = 0
> read(3, "TZif2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\v\0\0\0\v\0\0\0\0"..., 4096) = 1707
> close(3)                                = 0
> mmap2(NULL, 397312, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb1313000
> munmap(0xb1313000, 397312)              = 0
> clock_gettime(CLOCK_PROCESS_CPUTIME_ID, {tv_sec=0, tv_nsec=238988501}) = 0
> gettimeofday({tv_sec=1600536483, tv_usec=623091}, NULL) = 0
> brk(0x3e0f000)                          = 0x3e0f000
> clock_gettime(CLOCK_PROCESS_CPUTIME_ID, {tv_sec=0, tv_nsec=239269502}) = 0
> clock_gettime(CLOCK_PROCESS_CPUTIME_ID, {tv_sec=0, tv_nsec=239505187}) = 0
> brk(0x3e6f000)                          = 0x3e6f000
> gettimeofday({tv_sec=1600536483, tv_usec=636126}, NULL) = 0
> stat64("/etc/localtime", {st_mode=S_IFREG|0644, st_size=2705, ...}) = 0
> gettimeofday({tv_sec=1600536483, tv_usec=640531}, NULL) = 0
> stat64("/etc/localtime", {st_mode=S_IFREG|0644, st_size=2705, ...}) = 0
> --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
> +++ killed by SIGSEGV +++
> Segmentation fault

supermerill, 2020-09-19:
I've has problem to make the command line working even on windows.
Didn't had the time to investigate it until now.

Pavulon87, 2020-09-19:
Funny, I haven't checked this on Windows before 😄

supermerill, 2020-09-19:
I found a bug, you can pull master to test.
Note that it shouldn't occur if dont_arrange is set at true.

Pavulon87, 2020-09-19:
> I found a bug, you can pull master to test.
> Note that it shouldn't occur if dont_arrange is set at true.

after adding **--dont-arrange**
`10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill
85 => Generating support material
88 => Generating skirt
90 => Exporting G-code to /home/pi/test.gco
Slicing result exported to /home/pi/test.gco`

Building from sources takes ages on rpi, but this is great improvement already 😃

Pavulon87, 2020-09-20:
Now it looks fine.
I have one more question. The problem is rather system or octoprint side.
I can run `"/usr/local/bin/superslicer" --slice -o "/tmp/tmpgdtrt_7z.gco" "/home/pi/.octoprint/uploads/cube.stl"` from ssh and it works great.
I can run python script `p = sarge.run('"/usr/local/bin/superslicer" --slice -o "/tmp/tmpgdtrt_7z.gco" "/home/pi/.octoprint/uploads/cube.stl"', stdout=sarge.Capture(), stderr=sarge.Capture())` and it works great.
But the same code in octoprint plugin cause error:
`An error occured while setting up locale.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.`
I know that this is part of **PrusaSlicer.cpp** `boost::nowide::nowide_filesystem();`

supermerill, 2020-09-20:
maybe a system variable is missing in the "octoprint env"; something like $LC_CTYPE or  $LC_ALL

https://www.boost.org/doc/libs/1_48_0/libs/locale/doc/html/locale_gen.html

If you can't set them up, maybe it's possible to set them on a default value at program startup....

Pavulon87, 2020-09-21:
`sudo systemctl set-environment LC_ALL=en_US.UTF-8`
Problem solved.

supermerill, 2020-09-21:
If you can write a little how-to, i'll add it to the compile doc.

Pavulon87, 2020-09-22:
I'm not sure if I can help. This was at least second attempt to the compiling at raspberry pi. The first time I ended up receiving a Segmentation fault. This time, as I wrote at the beginning, it was similar, but I installed a lot more additional libraries in hopes of improvement.

I think that the most important thing is preparing Raspberry Pi.  I'm using overclocked to 2GHz RPi 4B 2GB with active cooling(it starts when the temperature is too high). 2GB of RAM is not enough and I turned swap on(+2GB), but i use SSD instead of memory card so it's not that bad. Swap is only needed during compilation, built-in memory is enough to run the GUI.

Build.PL from this repo + maybe this can help solving few dependencies:
`sudo apt-get install -y git cmake libboost-dev libboost-regex-dev libboost-filesystem-dev libboost-thread-dev libboost-log-dev libboost-locale-dev libcurl4-openssl-dev libwxgtk3.0-dev build-essential pkg-config libtbb-dev zlib1g-dev libcereal-dev libeigen3-dev libnlopt-cxx-dev libudev-dev libopenvdb-dev libboost-iostreams-dev libnlopt-dev libdbus-1-dev`

Running cmake will give information about the remaining missing dependencies.

I used:
`cmake .. -DCMAKE_BUILD_TYPE = Release -DSLIC3R_BUILD_TESTS = OFF -DSLIC3R_GTK = 2`
But other settings also work, such as:
`cmake ..  -DSLIC3R_STATIC=0   -DCMAKE_BUILD_TYPE=Debug   -DSLIC3R_FHS=1   -DSLIC3R_GTK=2   -DSLIC3R_PCH=0   -DCMAKE_INSTALL_PREFIX=/usr   -DCMAKE_INSTALL_LIBDIR=lib`

next:
`make -j3` or `make -j4` to use 3 or 4 cores

I think it is a good idea to run this command in [screen](https://linux.die.net/man/1/screen), because it takes a long time to complete and the connection may time out. [screen](https://linux.die.net/man/1/screen) will remain running in the background.

Example:
> screen -S superclicer
> #new terminal will open
> make -j4

and after disconnect:
> screen -x

Finally, it remains to run `sudo make install`

May also be useful to run
> sudo systemctl set-environment LC_ALL=en_US.UTF-8

I'm sure I forgot something, but I would have to start the installation from scratch to check everything.

ETE-Design, 2020-11-22:
This seem's interesting... World be great with an image og sine kind so you not have to Compile it :-) Do you need the complete Raspbian GUI or is it possible to use it in the Lite, and then install GUI for SuperSlicer only?

Pavulon87, 2020-11-22:
There is interesting github repository for PrusaSlicer - https://github.com/davidk/PrusaSlicer-ARM.AppImage
I'm sure that someone could make something similar for SuperSlicer.

ETE-Design, 2020-11-22:
@Pavulon87 Your right on that... Would even be greater if someone with the knowlege could build a "Complete" Image form Raspberry Pi OS Lite only with minimum dependencies to save memory usage...

supermerill, 2020-11-22:
There's no raspbian cloud image on the github actions, so for it I have to create a self-hosted one on one of my raspberrypi.

ETE-Design, 2020-11-22:
@supermerill Would be great if you could do... But then again to keep the OS up to date you have to merge newest version of Raspberry PI OS with your newest SuperSlicer...
How long does it take to compile a Image on an Raspberry pi, cause Nightly would of cause be great :-)

supermerill, 2020-11-22:
don't know, maybe a day

closed 2020-11-25 by supermerill

MrRKernelPanic, 2021-11-07:
@Pavulon87 I tried above steps, tried both using the BuildLinux.sh script, it seems to get most of the way through, but will not produce the AppImage as it seems to be generating this for X86 architecture.  I also tried using the standard make commands, again failed.

Finally I tried compiling with CMake using Ninja, this did the trick for me.

ksanislo, 2021-11-15:
@supermerill I can confirm that at least on aarch64 betas that after correcting the machine architecture in build_appimage.sh for which app tool is downloaded, the BuildLinux.sh script works as expected. It seems likely that is the only change needed to get RPi build support working for everyone.

supermerill, 2021-11-15:
@ksanislo 
can you drop here the modified script? so i can ad a rpi option to the main one.