supermerill/SuperSlicer #487: Klipper Mode does support progress bar on Ender 3

madsi1m, 2020-09-20:
### Version
2.20.53

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Ender 3 + SKR mini + Klipper

### Behavior
When "g-code flavor" is set to klipper the progress bar on my lcd is always at 0%
Perhaps progress isn't put in the gcode?

supermerill, 2020-09-20:
Probably.
i'll look to add M73 also to klipper

1N4148, 2020-09-22:
M73 are already written in the gcode when using klipper flavoured gcode. Just tick "Supports remaining times" in the printer settings dialog.

madsi1m, 2020-09-22:
Hi @1N4148 thanks for the info, that did the trick, can probably close the issue.

In PrusaSlicer+Marlin, this tickbox triggers the display of time remaining rather than progress bar. And the progress bar is always "on".

closed 2021-04-27 by supermerill