supermerill/SuperSlicer #491: Move filament_max_volumetric_speed to extruder

supermerill, 2020-09-23:
This value is more dependant on the extruder than the filament.
I should add a max_volumetric_speed_percentage into filament (so we can set 50% for abs, if it's harder to heat)

first discussed on issue #484
Discussion about this feature should continue here.

yschroeder, 2020-09-23:
Don't forget flexible materials, they also need very low volumetric speeds.

Maybe the input field in the filament should allow percentage of the extruder and absolute values. In case an absolute value is configured, the minimum of extruder speed and filament speed is used.

supermerill, 2020-09-23:
So
* I leave the current filament max_volumetric_speed
* I add a max_extruder_volumetric_speed in the extruder
* I add the possibility for the filament to enter a % (of the extruder)

The extruder max_extruder_volumetric_speed isn't used anywhere but to calculate the filament max_volumetric_speed

neophyl, 2020-09-23:
I'd disagree, the biggest single variable for max volumetric is the filament type,  It can already be set in both Print and filament profiles using the lowest value in either and given that you can limit the print or filament profiles to only work with specific printers, ie extruders then do we really need it to be more complicated ?

supermerill, 2020-09-23:
> do we really need it to be more complicated ?

there are printer with multiple toolheads (exemple: normal e3Dv6 0.4mm and with volcano0.6).
How to adapt the max volumetric to make the volcano print faster?

kursatu, 2020-09-24:
I would prefer this, not sure if it is different than what's been discussed. 
It is not simple though.
1- Leave the current filament max_volumetric_speed
2- Add a max_extruder_volumetric_speed in the extruder
3- Add the possibility for the **print_max_volumetric_speed** to be a % (of the extruder)
Please note the print here.

I am using Dragon hotend, normal flow (40mm^/sec) and high flow (60mm^3/sec), interchangeably and 
printing with volumetric flow rate.
I've always felt the need to define the max_extruder_volumetric_speed.
For certain filaments, I do set set filament_max_volumetric_speed because they just can't print that fast. Eg. polymaker/polysmooth.

neophyl, 2020-09-24:
Kursatu's suggestion sounds pretty logical.  So define a setting as part of the extruder hardware as a baseline and then modify the limit based on the filament being used.  
Would it be better/worse to have the setting under filament be just a % of the extruder value or both % and actual limit value ?

supermerill, 2020-09-24:
The print_max_volumetric_speed is only used if you set all your speed to 0 and let the auto-speed kick in.
It's not a "print_max_volumetric_speed " but more of a "base speed/flow for auto speed".

yschroeder, 2020-09-24:
I never used the `print_max_volumetric_speed`, I rather set all speeds to my travel speed and then limit the print speed via the filament `max_volumetric_speed`. This technique is also discussed in this video: https://www.youtube.com/watch?v=_bt1UZAnxnA

This has the additional advantage, that I can reduce the speed for external perimeters for better surface finish while printing everything else as fast as possible.

I would be fine if the `print_max_volumetric_speed` would impose the same type of limit as the filament  `max_volumetric_speed`. That way it would be possible to define printing profiles that give better surface finish by limiting the volumetric speed in the print profile.

Basically this:
* `max_extruder_volumetric_speed` defines what the extruder can handle in mm³/s
* `max_filament_volumetric_speed` defines an upper limit for the filament (e.g. for flexible materials), can be as % of `max_extruder_volumetric_speed` 
* `max_print_volumetric_speed` defines the target speed for the printing profile (to adjust surface finish)

The slicer uses the lowest volumetric speed from all 3 provided parameters.

* the non volumetric speeds can be set to 0 to use the highest possible speed based on volumetric speed with the travel speed as an upper limit

The auto speed feature should be removed, as it serves as an alternative to linear advance/pressure advance by trying to keep the extruder pressure constant. The auto speed volumetric speed parameter is misleading in conjunction with the other two volumetric speeds.

supermerill, 2020-09-24:
> The auto speed feature should be removed, as it serves as an alternative to linear advance/pressure advance by trying to keep the extruder pressure constant. The auto speed volumetric speed parameter is misleading in conjunction with the other two volumetric speeds.

No, pressure advance is made to reduce the pressure when the speed varies (deceleration / acceleration at a corner for exemple). It's made to be simulate an infinite acceleration. But i don't think it will correct the pressure for varying flow.
The auto-speed try to set a constant flow (but won't do anything for acceleration).

And most people don't have access to a (good) pressure advance printer (yet) anyway.

pemerick07, 2020-12-05:
I like this idea because it allows different printer settings to be created depending on the extruder and nozzle configuration. This is where the nozzle size is set which has a big impact on the MVS value. Then the filament configuration can be scaled based on the printer settings.

I'm running into a problem with needing a different max volumetric speed for my support filament (extruder 2) and my primary filament (extruder 1) because they are different extruders, hotends and nozzle sizes. The primary filament MVS is being enforced, but the support MVS is not. I'm assuming that if the MVS was set per extruder in the printer settings, I set each filament as a percentage of the extruder MVS and I set extruder 2 as my support extruder in print settings, this functionality would work correctly.

kdb424, 2022-02-04:
This would be massively helpful to have for printers that change nozzle sizes often, like a revo system. A .8 nozzle will have much different flow than a .4 nozzle. Moving this to the extruder section may be best as pemerick07 said as IDEX printers or tool changers can have different nozzles, and therefor flow rates. This would allow you to set a global volumetric max for each extruder, and tune the filament (with percentage) to the extruders, making for a very versatile setup to get the most out of your hotend, without having to worry about outrunning it.