supermerill/SuperSlicer #493: Very thick line width for overhangs

kursatu, 2020-09-24:
### Version
_Version of SupserSlicer used goes here_
Latest code from github, 9 September, 2020
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
CoreXY printer , Vking derivative
Marlin bugfix-20.x
### Behavior
* _Describe the problem_
The overhang detection is one of the most important functionality of SuperSlicer, imo.
But it over extrudes for some reason.
It became very visible with 0.8mm. I was trying to fix 
this problem with 0.4mm nozzle and always though that it is some other printing problem.

* _Steps needed to reproduce the problem_
- Use the attached 3nf file.
- Slice. See the overhang being really thigh.
- Change: "Width and Flow"-Flow Ratio-Bridge to be 10%, then you can get some reasonable quality print.
Needles to say, this workaround is not really usable in general.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
[image]
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_
[Defect9.zip](https://github.com/supermerill/SuperSlicer/files/5273166/Defect9.zip)

neophyl, 2020-09-24:
Its printing as a bridge and if I recall correctly a bridge always prints as the nozzle diameter so its producing a section that is 0.8mm wide and high, hence the very thick layer.  For this particular model if you turn off -  Perimeters & Shell>Overhangs>On Perimeters then it slices it correctly I think.

supermerill, 2020-09-24:
like neophyl said, you can't something flat if there is nothing to push against.

TD5023, 2020-09-24:
@supermerill Would it be possible to create a new setting where the overhang threshold is lower and doesn't include bridge flow? It would be awesome to be able to apply bridge speed and cooling to shallower overhangs while still using regular layer flow.

kursatu, 2020-09-24:
@supermerill  I disagree that this is working as intended. 
You can't have such massive over-extrusion, ie. width and height equal to nozzle diameter, from sources. 
It will make them sag even more.
In fact it prints a lot better , if you don't use overhang detection and slow down print speed for better cooling
The overhangs should probably always print with regular layer flow, certainly not with bridge flow.
This is pretty much the only essential feature that made me use superslicer.

robthide37, 2020-09-24:
doesnt turning off as bridge fix your prob tho ? i def noticed it on a benchy before but i am just asking why its not ok to just turn it off ?

supermerill, 2020-09-25:
@TD5023 it's possible, but i need some experimentation result to show how useful it can be.
Print a test piece with the good parameters for the overhangs, and an other one with the good settings for the bridge. By looking at them, we must be able to see one with good overhangs and bad bridge and vice-versa.

@kursatu 

> In fact it prints perfectly, if you [...]  decrease the bridge flow a lot

So do it. The settings exist in bridge flow.
If it's not good, then same thing as for TD5023, for this kind of radical change on overhangs I need some test results to support my work.

TD5023, 2020-09-25:
@supermerill I made a little test piece awhile back that I just printed three ways at .15 mm layer height. They are as follows (direction reversal is active in all of them):

1) default (standard layer settings with no overhang adjustment)
[image]
On this one, the straight lines are perfectly fine, but the surface is rough in rounding the corner. Cooling is an issue here.

2) as a bridge
[image]
This suffers from obvious over-extrusion and is not suitable for shallower overhangs (which I'm sure you're already aware of, but I wanted to be thorough).

3) with the outside perimeter speed and cooling manually set to mirror my bridge settings, but standard layer flow
[image]
This one turned out nearly flawlessly. I have no complaints.

Hopefully my crappy photography adequately conveys what I'm describing. This model is set to a 40 degree overhang (from horizontal), so it's decent, but not gigantic. At my layer settings, that amounts to a little more than 50% perimeter overlay between layers (so it wouldn't even register as an overhang on PS). The best print was definitely the one with the bridge speed and cooling, so if there was a way to trigger that on any overhang without having to default to these settings on all external perimeters, I'd greatly appreciate it.

Also, as an extra note, would it be possible to turn off an "as bridge" trigger on layer two only? Elephant foot compensation causes it to activate, which results in over-extrusion. I just noticed this while doing these tests.

kursatu, 2020-09-26:
Below is something I tend to use for testing. It is small and prints fast enough.
I have an older print, after the previous overhang defect was fixed. 
It shows over-extrusion, though I did not know it at that time.
I'll print another one soon once I fix my build. 
I was trying to change the code to use layer flow (instead of thick bridge flow) for overhang perimeters.
Can post the older prints if needed, probably not very informative after TD5023 's findings.
https://www.thingiverse.com/thing:2975429

robthide37, 2020-09-27:
@kursatu why would you give a thumbs down for me asking a question to try try and understand better and possibly help you . Also you thumbs down merril and than expect help when he is explaining that its working as he intended . Guess what its his fork so if he intended full nozzle width then its working as intended and you are being immature. If you want something diff you just have to explain. I was asking you to clarify what you wanted on without the as bridge feature so you simply need the settings for cooling divorced from the extrusion modification. Could been nicer about it and he prob would have already done it for you .

supermerill, 2020-09-28:
@robthide37 you can also see +1 -1 as "vote"
What is the need for -1 if it's rude/forbidden?

kursatu, 2020-09-28:
@TD5023 If you have your own build you can apply the changes at #501 and see how it works for you.
I print a wide variety of stuff and also for actual production.
3D printing is an indispensable production method for me.
Therefore, I've decided to also start using Cura, after many years. 
It has the features that I need, like overhang speed & cooling, 
avoid crossing perimeters during travel etc. 
I had to set it up for multimaterial , which was not easy to do.

Just to recap, this defect is due to chose for an incorrect behavior and the consequences are:

1- Mess up the dimensional accuracy by extruding 3x-7x more around where the problems 
may occur. It will guarantee the sagging due to excess material.
2- Increase the print failures because of increased curling due to excess material. 
The nozzle hits the curled sections during travel. It is amplified with the other defect. 
3- It is very hard to notice and therefore impossible to correct. It happens only in a small area 
and confuses you as to why the problem 2 might happen.

I now realize, all these caused me to loose huge amount of time.
I've filed several defects for overhangs, out of around 10 total defects.

TD5023, 2020-09-28:
@kursatu I would, but I'm not much of a programmer. For perspective, when I need to write a program for straightforward calculations and outputs, I use Fortran. As a result, I don't even try digging into these and risk messing anything up. 

As for Cura, I never liked its interface. It's a good slicer, but I've always preferred Slic3r and its forks. General overhang control is the main thing I've always found to be lacking, but other than that and a few fairly inconsequential changes I'd like to see, it's been good for me.

robthide37, 2020-10-03:
> @robthide37 you can also see +1 -1 as "vote"
> What is the need for -1 if it's rude/forbidden?

your right i guess i always took it as if incorrect or rude than you downvote so asking question didn't make sense to be downvoted to me but guess i took the wrong way. Anyway as much as the feature works the way you intend do you have any plans to make a divorced setting where you can not alter the extrusion but still use the higher fan speed ?

supermerill, 2020-10-04:
i'm leaning toward replacing the 
 * as bridge checkbox
 * as bridge threshold

by

* bridge flow threshold
* bridge fan & speed threshold

robthide37, 2020-10-04:
> i'm leaning toward replacing the
> 
> * as bridge checkbox
> * as bridge threshold
> 
> by
> 
> * overhang threshold (for fan & speed)
> * bridge flow threshold
> 
> but do I apply the bridge fan & bridge speed if the bridge flow is applied? I think I should. But maybe not.

I would either just keep simple where it uses bridge fan and speed when the overhang threshold is reached like you said with a seperate bridge flow threshold or maybe just add a seperate overhang flow and fan setting or really just an overhang fan setting and then the checkbox would just be for bridge flow but now on the cooling page would be an additional fan speed for overhang . I would imo just implement the simplest solution as far as coding since they all should work

kursatu, 2020-10-05:
I would just change the wording of the options to 
- Use bridge speed and fan 
- Overhang threshold
And **always** use the perimeter flow for the overhangs. 
It would convey the correct message and most importantly correct the over extrusion
The current over extrusion only decreases the overhang quality and increases print failures.

TD5023, 2020-10-05:
> I would most importantly, **always** use the perimeter flow for the overhangs.
> The current over extrusion only decreases the overhang quality and increases print failures.
> Also, I would change just the wording of the options to
> 
> * Use bridge speed and fan
> * Overhang threshold
>   It would convey the correct message.

I'm in agreement with this. The only time I could see bridge flow being used properly for an overhang is if it's a simple straight line between two anchored ends and the exterior perimeter is entirely in air. However, at that point, it should be actually treated as a bridge, since that's what it is in application. However, for "normal" overhangs, even very steep ones (like the top of a circular hole), regular perimeter flow should pretty much always result in a cleaner print. I'd be happy to print tests of both if it would be helpful for comparison.

kursatu, 2020-10-05:
@TD5023 I think I can produce you a release for testing, if you are using Windows. It may contain the things I wrote above, certainly the fixes for the 2 defects. It wont be able to run the test suite on it though, because I don't know how.

supermerill, 2020-10-05:
> I'm in agreement with this. The only time I could see bridge flow being used properly for an overhang is if it's a simple straight line between two anchored ends and the exterior perimeter is entirely in air.

55° to 85°, with bridge flow (and alterning direction).
[image]

TD5023, 2020-10-05:
@supermerill what layer height are you using for that? It looks pretty thick, though that could be a trick of the picture. I use adaptive heights for everything, so steeper overhangs will be printed at or close to my minimum, which is .04 mm. Not sure if that would make a difference (I admit I don't know the math that goes into bridge flow, just that I get overextrusion/drooping when it's active).

@kursatu I'll take a look at that later today.

kursatu, 2020-10-05:
@TD5023 Here is a link to the test model of the other defect. You can slice and dice it to get only the overhangs section. Otherwise, it can take longer to print.
https://github.com/supermerill/SuperSlicer/issues/501#issuecomment-703780938

robthide37, 2020-10-06:
Even the benchy prints better with the old overhang method off and i am testing your build right now comparing to ideamaker cause i have had to use ideamaker for my maf housing and some adapters since the overhangs need the cooling but on the arch it causes the sensor to not fit without cleanup post print which costs alot of time when printing batches of 20

kursatu, 2020-10-06:
@TD5023 @robthide37 I've just really published the release. It was mistakenly in draft mode and I did not notice. Sorry for the confusion.
Here is the link, it is visible now.
https://github.com/kursatu/SuperSlicer/releases/tag/V1

robthide37, 2020-10-06:
was trying figure out i updated my visual studio and now it can never find boost when doing c its driving me nuts i may reformat and start over so thanks for linking the compiled one

TD5023, 2020-10-07:
I printed the test model both ways (no supports at all). Here are my results:

[image]
This is the original with bridge parameters set to activate at 50% (the default)

[image]
This is with the new version that does not adjust flow and remaining bridge parameters set to activate at 25%.

[image]
Here is a side-by-side. 

It's most evident in the last picture, but the original version is significantly rougher once the bridging kicks in. My cooling setup isn't the best and I definitely need supports for extreme overhangs, but in the zone of 45 degrees to about 70, just cooling and speed were the best.

robthide37, 2020-10-07:
voron cube and benchy also way better with kursatu's build

kursatu, 2020-10-08:
So, the improvements are verified. 
Can we please get this behavior changed and both defects committed?
This is an important feature and it should be very beneficial to everybody.
PS: I am expecting even more quality difference in comparison to TD5023's tests.
My cooling is pretty good so the only problem with overhangs were the over extrusion.
My machine is busy so I can't provide a before & after comparison at this time.
But here is an earlier print, which shows the over extrusion starting at 45 degrees and sagging at the very top.
Such pictures are not easy to take, so it is not easy to see.
[image]

supermerill, 2020-10-08:
> Can we please get this behavior changed and both defects committed?

yes.

@TD5023 0.2mm layer height on 0.4 nozzle I think, that's my default.

robthide37, 2020-10-30:
If you set the bridge flow to 0 to disable then it treats all perimeters as bridges and sets fan to the bridge speed for the entire print

robthide37, 2020-10-30:
[fan_test_-_Part_2b.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5462870/fan_test_-_Part_2b.3mf.txt)

robthide37, 2020-10-30:
ps the kursatu build was working idk if it is a prusa merge issue but also make sure to set fan startup delay to 0 from the default-1 as that reeks havok when at -1

supermerill, 2020-10-30:
@robthide37 if ot's a bug, you should create a new issue, as i can't close/tag this one for that bug.

closed 2022-04-01 by supermerill