supermerill/SuperSlicer #494: Vase mode 'skipping' layers

ProfEngr, 2020-09-24:
### Version
2.2.53

### Operating system type + version
Win 7

### Behavior

Seems like I'm doing something wrong here or I found a bug. I'm trying to slice the vase mode pyramid from Thingiverse and I'm not getting a printable result. The image as an example of what happens regularly throughout the preview. The slicer generates 3 more empty, square perimeters (a 1.0mm void) before continuing the pattern. It happens everywhere you see the thick lines in the image.
I also noticed that, while I have 1 bottom layer selected, the slicer is generating 2 filled layers before starting the vase path.

[image]

This is how S3D slices the same section. Hopefully it's easy to see the pattern filling the interior.

[image]

#### Project File (.3MF) where problem occurs
I saved a 3mf, but I have my API key stored in the Printer tab. If that gets exported then I can't upload the file. Here's the STL.

[pyramid6.zip](https://github.com/supermerill/SuperSlicer/files/5276794/pyramid6.zip)

supermerill, 2020-09-24:
did you set the slice_closing_radius to 0 (slicing tab) ?

ProfEngr, 2020-09-24:
Thanks, that fixed the issue of the skipped layers. Now I'm stepping through the gcode line by line again. It seems the planner decided that each square should be drawn independent of the previous instead of 1 continuous extrusion. Did I miss another setting?

supermerill, 2020-09-24:
try "overlapping perimeters" the walls may be too thin for your extruder.

ProfEngr, 2020-09-24:
Tried it both ways. It was already checked.
With:
[image]

Without:

[image]

Ugh. I managed to overwrite the S3D gcode I was using as a reference for a successful vase mode slice. Just perfect. I can't get S3D to duplicate its original slice paths now either.

ProfEngr, 2020-09-24:
What's interesting is that using the import from PS earlier this year most of the STL slices in a vase path. The first 2 'layers' are put down w/ travel moves then layer 3 starts moving as I would have expected the whole part to go. Here's the gcode for reference, but I have zero bottom layers set in the print tab (S3D has 1 bottom layer set). There are also a handful of layers further up the part that contain travel moves. I had to uncheck gap fill as well or that would pop in sporadically.
[pyramid6-ss-s3d-gcodes.zip](https://github.com/supermerill/SuperSlicer/files/5279486/pyramid6-ss-s3d-gcodes.zip)

Pavulon87, 2020-09-25:
There is also problem with retractions in vase mode.
[image]

ProfEngr, 2020-09-29:
Related issue #4805 that may offer some explanation to my original issue here. Also, extrusion width of 0.4 cleared up several problems; seems 0.48 wouldn't fit in the pattern spacing.

With the new STL referenced in 4805 I managed to get a proper slice in Prusaslicer. I haven't tried it with this fork yet.