supermerill/SuperSlicer #495: Placeholders for X_min / X_max / Y_min / Y_max?

CCS86, 2020-09-24:
Hi guys,

Are there placeholders built in for the min / max X and Y coordinates of the current sliced job? Even better would be specific to the first layer / footprint.

I'd really love to drive mesh leveling coordinates using these per print values for fast, efficient probing.

Thanks!

supermerill, 2020-09-24:
If gcode_label_objects is set (output -> label objects), then you'll ahve a json string in a comment in the header with bounding boxes.

CCS86, 2020-09-24:
Thanks for the quick reply!

That's not quite what I have in mind. Those object labels, just give per-object bounding box coordinates.

My goal is to build a mesh grid definition, which allows automatic probing of only the build platter area used in a given print. 

M557 X0:200 Y0:220

My hope was that with a placeholders to track the min/max coordinates, I could put a call like this into my start gcode:

M557 X[x_min]:[x_max] Y[y_min]:[y_max]

And let Superslicer populate the values for me.

CCS86, 2020-09-24:
I actually just got a workaround functioning. A friend helped me develop a Python script for Cura to parse the gcode for these values and replace a dummy M557 call in the start gcode. Another user adapted it for use in Slic3r.

Integrating placeholders into Superslicer would certainly be more flexible for different printers / firmware. but for now, it is working for me.

Here is that code in case you were interested in developing it and it was helpful.

[duet3d_automesh.py.txt](https://github.com/supermerill/SuperSlicer/files/5278024/duet3d_automesh.py.txt)

supermerill, 2020-09-30:
I'll add 
`; plater:{\"center\":[%f,%f,0],\"boundingbox_center\":[%f,%f,%f],\"boundingbox_size\":[%f,%f,%f]}`
after all object ones, and it will be the merged bounding box of all other.
This is good enough?
Maybe I'll need to change for a circle if the bed is round.. what do you think?

CCS86, 2020-09-30:
I'm not so experienced with round beds, so I'll defer to someone else.

My thought is that if the data is still just posted in a comment like that, you still need a post processing script to parse that data and format the mesh command. 

Is it hard to push that data into a new "placeholder"?

supermerill, 2020-10-01:
I'll look to it

closed 2020-11-25 by supermerill