supermerill/SuperSlicer #497: Filament change in Plater tab not reflected in Filament tab

carloshespanhol, 2020-09-24:
### Version
2.2.53

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
Hypercube / Klipper

### Behavior

When changes de filament in Plater tab it is not changed in Filament tab, but when clicks in the edit preset button it changes the selection in Filament tab. It could be confusing.

supermerill, 2020-09-24:
you use multiple extruders?

neophyl, 2020-09-25:
This is the same in Prusa Slicer too.  I helped a guy on the Prusa forum with this same issue, he had multiple extruders configured and couldn't understand why changes made to the filament settings weren't in his output g-code.  
With a single extruder its very simple as you can just apply filament tab settings directly to the extruder, and this is done.  If you are paying attention it even tells you as the filament preset shown on the plater dropdowns gets changed from say "ABS" to "ABS (Modified)".  

However when you have multiple extruders those filament dropdowns only show **saved** filament presets as you can not know which extruder the filament you are editing is to be applied to.  None will show "ABS (Modified)".  So the changes you made are never applied to the slice.

Yes its confusing if you don't know what is happening but I don't see a simple way of getting around this.  Possibly a message when you try and leave the filament tab if you have made changes prompting you to save and apply it (but only for multi extruders).  Normally I'm against pop ups so better ideas welcome :)

carloshespanhol, 2020-09-25:
> you use multiple extruders?
Yes, I have 2 extruders

> the problem is that i don't know what to show by default, the first filament from the list?
> 
> what is the better behaviour?
In my opinion, when you change one of the filament selects in Plater tab it should have the same behavior than when you click the edit preset button for that filament,  it would change the filament in the Filament tab and if it have any change it would ask if keep or discard the same way when you click the button. The only difference is that it dont change tabs.
This way we will never have the situation that happens today, when you can have a filament in Filament tab that is not being used in any extruder.

webeule, 2024-12-14:
I am just coming from Prusa Slicer and found it very confusing that my filament changes are not reflected in the plater.
In Prusa Slicer you have just a drop down where you can select the Extruder. I would assume that this will be the same in Super Slicer when it has been upgraded to the latest version of Prusa Slicer?