supermerill/SuperSlicer #498: Ironing creating gaps

afkfurby, 2020-09-25:
### Version
2.2.53.2

### Operating system type + version
win10

### Behavior
Tried to dial in ironing with petg on one of our big machines, nozzle size 0.8mm
but doesn't matter what I change, it's creating gaps betweend shells and solid infill of the ironing pass,
This does not appear when using rectilinier (filled).
It's also appearing with PLA Carbon.
Doesn't make a change if I use one or multiple shells on top layer.

[image]

#### Project File (.3MF) where problem occurs
[ironing.zip](https://github.com/supermerill/SuperSlicer/files/5282255/ironing.zip)

supermerill, 2020-09-25:
In the ironing infill, the first pass is done without overlap. (but not the second one).
Maybe that's why.
I did that to avoid overextrusion on narrow surfaces.
I'll have to change that, maybe it's possible to have an overlap on large areas but (almost) none in thin one...

robthide37, 2020-10-06:
one thing is when ironing is on it changes the top infill speed to the ironing speed and then your solid infill becomes your top layer speed

fishcu, 2020-11-23:
I've also come across this and would consider this a bug. I would prefer the overlap setting to apply everywhere.

Compare the following two screenshots that show top infill as "monotonous filled" vs. "ironing":

[image]
[image]

supermerill, 2020-11-23:
if you have "enforce 100% fill volume", you have the same amount of plastic on the two cases.
and i have done that to prevent overextrusion on thin area.

supermerill, 2022-03-23:
the behavior has changed in the last releases, with a bit more overlap.

closed 2022-04-01 by supermerill