supermerill/SuperSlicer #500: Seam in vase mode :|

cab404, 2020-09-26:
### Version
2.2.53.2

### Operating system type + version
NixOS

### 3D printer brand / version + firmware version (if known)
TronXY X5SA + Klipper

### Behavior
Vase mode generates strange acceleration shifts between layers, which cause visible seam to appear on finished prints
Example:
```
G1 Z0.998 X131.323 Y135.741 E238.55112 ;  perimeter external
G1 Z0.999 X133.004 Y133.919 E238.60043 ;  perimeter external
G1 Z1.000 X134.521 Y132.187 E238.64625 ;  perimeter external
G1 Z1.000 X134.760 Y131.939 E238.65310 ;  perimeter external
M106 S249.9 ; enable fan
G1 Z1.000 F4800.000 ; move to next layer (8)
G1 F2400.000
M106 S255 ; enable fan
G1 Z1.000 X135.492 Y131.307 E238.67227 ;  perimeter external
G1 Z1.001 X136.231 Y130.667 E238.69171 ;  perimeter external
G1 Z1.002 X138.605 Y128.861 E238.75107 ;  perimeter external
```

[image]

#### Project File (.3MF) where problem occurs
[tongue.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5287386/tongue.3mf.zip)

Pavulon87, 2020-09-27:
https://github.com/supermerill/SuperSlicer/issues/494#issuecomment-698890820
Strange things are happening.
[image]

cab404, 2020-09-27:
It is actually slices the same in PrusaSlicer.

cab404, 2020-09-27:
I will try removing all the acceleration changes and printing without those.

cab404, 2020-09-27:
I read a bit of SpiralVase.cpp. So, basically, post processing of layers leaves X and Y in place, and only changes Z. That approach basically tilts all the layers, and that generates artifacts. 

I'll try making vase mode which interpolates between X and Y positions on two layers, instead of only spiralizing one.

supermerill, 2020-09-28:
This can be difficul for certain geometries like the fractal pyramid.
You may be able to create counter-exemples for every algorithm you try.
But I pray that you can create something that wirks great, at least on common vases object, and we should be able to deactivate it (per layer) if needed anyway.