supermerill/SuperSlicer #501: Incorrect overhang speed

kursatu, 2020-09-27:
### Version
_Version of SupserSlicer used goes here_
The latest release from Github 2.2.53.2
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
The overhangs are not printed at the bridge speed.
The bridge speed is correct though.
If you look at the top left corner, you'll that it is printed sporadically with bridge speed.
* _Steps needed to reproduce the problem_
- Open the attached 3.mf file
- Slice 
- Check the speeds and the feature type
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
The overhangs should print at the bridge speed.

* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_
[Defect10.zip](https://github.com/supermerill/SuperSlicer/files/5287693/Defect10.zip)

kursatu, 2020-09-27:
Thanks for the good work. 
I am in the process of fixing it, for learning purposes.
It'll be slow. I'll create a PR if I can get it fast enough.
It might be trivial if you know where to look.

kursatu, 2020-09-28:
Fixed it. The minimum fix is
GCode.cpp: Line 3539
        gcode += this->_extrude(*path, description, path->role() != erOverhangPerimeter ? speed : m_config.get_abs_value("bridge_speed"));
Also fixed #493
The minimum fix is
LayerRegion.cpp: Line 84
    g.overhang_flow         = g.ext_perimeter_flow;

[image]

supermerill, 2020-10-01:
The speed should be set via _before_extrude on line ~4101
```
} else if (path.role() == erOverhangPerimeter || path.role() == erBridgeInfill) {
     speed = m_config.get_abs_value("bridge_speed");
}
```

kursatu, 2020-10-05:
The first model is a 3d printer test from the following link.
https://www.thingiverse.com/thing:2975429

supermerill, 2020-11-15:
seems fixed in the latest nightly

kursatu, 2020-11-15:
I've pulled and built the latest sources and both .3mf files slice as expected.

closed 2020-11-15 by kursatu