supermerill/SuperSlicer #502: Doesn't repair STL-Files

Hopser2k, 2020-09-27:
### Version
2.2.53

### Operating system type + version
Linux Mint 20

### 3D printer brand / version + firmware version (if known)
Tronxy XY-2Pro with newest Marlin Color_UI, newest, because it changed.

### Behavior
* _Describe the problem_
SuperSlicer doesn't repair STL-Files. The repaired OBJ-Files has the same Problems as the origin STL-Files

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_

neophyl, 2020-09-27:
Could you provide an example for the developer to at least examine ?  Some files just don't get fixed by any automated process and need the human eye to fix.

tweiner, 2020-09-29:
I will send an example.  But the issue is when the program detects a non conforming part that needs to be fixed the triangle with exclamation pint.  Unlike when you click on it in Prusaslicer it goes out and fixes it.  Superslicer does not.
[image]

tweiner, 2020-09-30:
here is a file that has a problem.

[tony weiner.zip](https://github.com/supermerill/SuperSlicer/files/5302558/tony.weiner.zip)

Hopser2k, 2020-09-30:
Hi,
Attached two STL-files that SuperSlicer will not repair.
[STL-files.zip](https://github.com/supermerill/SuperSlicer/files/5303301/STL-files.zip)

tweiner, 2020-10-07:
There have been multiple files uploaded.

neophyl, 2020-10-07:
Does Prusa have an api key that allows them to use Netfabb?

closed 2021-06-14 by supermerill