supermerill/SuperSlicer #503: Filament specific print speed overrides

Exerqtor, 2020-09-27:
How does really this work? I'm having a hard time(putting it lightly) tunning my printers profile when it comes to different types 
of filaments. For instance if i tune it for an PLA filament and get that nailed down. And then try to setup a PETG or CPE filament i would

supermerill, 2020-09-28:
Vol speed is a hard limit for melting power for your extruder/fikament combination.
The slicer ensure you never ecceed this. There is currently no direct print speed override for filament.

Exerqtor, 2020-09-28:
That's what i thought, wouldn't a filament max speed override / speed multiplier be a nice approach to adjusting the speed depending on the filament? So several print profiles can be avoided. For instance using a profile setup for PLA (since it's usually the filament people usually are able to print the fastest with) as a baseline with it's speed-multiplier set to 1. And if you swap to PETG you just adjust the speed-multiplier to 0.5 and end up with a 50% speed reduction.

1N4148, 2020-09-28:
Wouldn't do a M220 <feedrate in %> in the filament specific gcode that?

Exerqtor, 2020-09-28:
That would mess with the travel, retraction and z-lift speeds to tho.

1N4148, 2020-09-28:
Ok, then it's no solution. I just use the max volumetric flow for these scenarios to scale down the topspeed

nhopfen, 2020-09-28:
I was really happy already with the print speed override in stock prusaslicer, having PLA at 15mm3, PETG at 8mm3 and tpu at 4mm3 on my D.D. ender 3. 
heat your nozzle at desired temps for filament and test with pronterface until you reach the max for your machine before it skips.
(manufacturers often offer values that can be used to calculate a rough maxmm3).

Then tune your speeds according to what you judge acceptable for ringing, and you should then be plug and play for any filament
There should be absolutely no need for further tuning after that.

Exerqtor, 2020-10-06:
Just checked out the new version with the Max print speed, and i see that it's displayed with mm3/s, that's a typo right?
Aaand this will just cap the max speed (judging by the name), and not work like a multiplier of sorts? So if a print profile is set to lets say 120mm/s default perimiter speed and 60mm/s for external, and the "filament max speed" set to  60mm/ both will just print at 60mm/s? Or will it be factored that 60mm/s is now the "new" default perimiter speed" and the exisitng % difference between all the other speed settings will be adjusted accordingly?

supermerill, 2020-10-06:
oups, yes, it's a cap in mm/s.

mmm, adding an override for the default speed in the filament tab may be a good idea...
the volumetric one can also override the one for the auto-speed...

1N4148, 2020-10-07:
Two speed limits per filament? Imho a little bit confusing cause the max. volumetric speed is doing the same. It would really make more sense to implement a speed multiplier instead of a 2nd speed limit expressed in mm/s. Just my 2 cents.

Exerqtor, 2020-10-07:
Yeah a max volumetric speed AND a speed multiplier (without the newly added max speed, since max mm3/s allready limits that) is the route I see being the most intuitive and best fix for the issue so to say. 

And add a wizzard for tuning the max mm3/s, so that it's easier to nail that down. 

And down the road maybe a wizzard to get the speed multiplier somewhat dialed in to (if it's even needed). 

But hey, ive been wrong before lmao.

1N4148, 2020-10-07:
Speed multiplier would be great. For now max. filament speed and max. volumetric speed are only a top limit.

nhopfen, 2020-10-07:
I still do not understand anyones point here.

For example, let's say you have set outer perimeter to 30mm/s with PLA because that is your personal limit before ringing/ghosting starts to appear and bother you, but have the other speeds set at 60mm/s.  Then you switch to a brand of TPU that extrudes well only up to 30mm/s, is there ANY REASON WHATSOEVER you would need the outer perimeter to run at 15mm/s instead of having the whole print limited to 30mm/s? where is the advantage of this?

1N4148, 2020-10-07:
You're getting a better quality for surfaces our outer perimeters if you stay a bit below the max. volumetric limit for this filament. I normally print non visual elements at full volumetric speed (e.g. 12mm³/s for PLA) and all visible elements (outer perimeters, top surface) at half speed. This rule is also valid for critical filaments like PA or PC. Cause of the hotend, there's the limit's about 3-4mm³/s. But for optics these half to be halfed...

1N4148, 2020-10-07:
Imho the whole speeds are more or less screwed up when trying to use autospeed. If I'm setting default print speed to 0 (for autospeed), all other speeds cannot expressed by percentages of the default speed. If this would be working we won't need a filament factor as it's already there by the max. vol. speed.

closed 2022-03-23 by supermerill