supermerill/SuperSlicer #505: [Feature] Start Octoprint job from SD Card

cab404, 2020-09-27:
On Klipper, there's an option to use gcode directly from filesystem, bypassing serial-to-klippy connection. Klippy emulates an SD card with it's contents mirroring those of Octoprint uploads folder.

SuperSlicer has an option to start print job right after upload, but does not allow to select SD card as a source of a file. That would be a nice printer option to have.

supermerill, 2020-09-28:
Looked on octoprint, seems easy, almost just a checkbox to add to the dialog.

cab404, 2020-09-28:
I would probably want it as an printer option. It's only applicable to klipper, and other printers would only suffer from it)

cab404, 2020-09-28:
Just to clarify — Octoprint should be commanded to upload to it's storage, but subsequent command to print should be from SD card.

supermerill, 2020-09-28:
Was thinking of a "to sdcard" checkbox near "start now".

cab404, 2020-09-28:
> Was thinking of a "to sdcard" checkbox near "start now".
> Save in sdcard. Start from sdcard.
> 
> There are uses outside of klipper.

Yeah, but it's a different feature I was asking for)
Cura implemented "Save to SD Card" with Octoprint, but as everything through serial port, it works atrociously slow.

supermerill, 2020-09-28:
I think it's exactly what you ask.
Or i don't know how to do it.

supermerill, 2020-09-28:
But how to write in the filesystem without going through the sdcard thing from octoprint?

cab404, 2020-09-28:
> But how to write in the filesystem without going through the sdcard thing from octoprint?

There is no way. I am just stating that it's slow, but sometimes useful.

cab404, 2020-09-28:
Virtual SD card is actually a folder in filesystem, which mirrors Octoprint's upload folder.

supermerill, 2020-09-28:
So if the sfcard checkbox is xhecked and the firmware in klipper, i have to upload normally but start from sdcard?

cab404, 2020-09-28:
If the checkbox is in the upload dialog, then yes. But I wouldn't want to use "upload to SD card" with any marlin printer, and would have to uncheck it back then using other octoprint instances for other printers.

That's why I would want it in printer settings)