supermerill/SuperSlicer #506: Feature Request: Slic3r's Improved 3D Honeycomb Fill

unicornswag, 2020-09-28:
In early 2018, Slic3r's 3D honeycomb infill pattern was updated to only alternate the print direction after the current layer of cells is completed, as opposed to changing direction every layer. This results in a huge increase in compression strength with low fill densities, since now only a small fraction of the toolpaths are bridges.

https://github.com/slic3r/Slic3r/commit/54fe2e4145a75fc7e4dddf24986fd6727ea06283

I made  a little write-up on the advantages [here.](https://forum.prusaprinters.org/forum/prusaslicer/prusaslicers-3d-honeycomb-infill-is-sub-optimal/) The improvement in compression strength is really dramatic, so I think it would be awesome to see this style of 3D honeycomb in SuperSlicer, alongside the standard 3D honeycomb.

supermerill, 2020-09-28:
Is there any advantage for the current 3d honeycomb implementation?

supermerill, 2020-09-28:
I'll see if i can't make it works better flow-wise with the old algorithm.

unicornswag, 2020-09-28:
I honestly don't see any real advantage in the current PrusaSlicer / SuperSlicer implementation of 3D honeycomb. In my testing, the Slic3r version produces much stronger parts in every direction. It's also a more "open-cell" design, so there's the possibility of using a filler material, such as expanding foam or plaster, to fill the cells after printing.

supermerill, 2020-09-28:
For open cell, the gyroid is here. Indeed, the updated pattern looks like a sub optimal gyroid.
With cubic and gyroid here, the 3dhoneycomp doesn't have any advantage that i can think of. The last use may be for art, and the enclosed cell looks better imho.

unicornswag, 2020-09-28:
I've been testing gyroid alongside 3D honeycomb recently, and Slic3r's updated 3D honeycomb is far stronger than gyroid at the same density. I've been building test parts between 2.5% and 5% fill density, and it's astonishing how much stronger the 3D honeycomb parts are. I think the main issue with gyroid is that the intersections between layers of cells form acute angles between toolpaths that easily split apart when loaded in compression in the X- or Y-axes. Slic3r's 3D honeycomb really does have a practical advantage in large, sparse-filled parts.
[image]

supermerill, 2020-09-28:
Yes, from cnckitchen test, gyroid is more isotropic, and others like stars are more strong in z.

unicornswag, 2020-09-28:
I don't think CNCKitchen ever tested gyroid against Slic3r's updated 3D honeycomb, but rather the PrusaSlicer version.

A major downside of cubic and stars is that, for a given density, they produce relatively large cells that need to be bridged over by the top solid layers. At the same density, the cells can often be twice as wide as with 3D honeycomb.

supermerill, 2020-09-28:
Be careful with the density setting, it's not really cimparabke from infill to infill. I tried to tune gyroid initially, but even this may have diverge a bit over changes.

Fot he cell size problem, there is a denser infill setting to solve that. Rectilinea rshould produce the tinyest cells and be strong in z.

unicornswag, 2020-09-28:
Yes, I generally don't use the "fill density" parameter as an actual metric. Rather, I'll slice a part and look at the actual length of filament used.

Rectilinear has decent compression strength in the Z, but extremely low tensile strength, since every toolpath is a bridge, with only one point of contact at each intersection.

supermerill, 2020-09-28:
How do you test for tensile test?
All my tensile test show that perimeters are the significant factor.

unicornswag, 2020-09-28:
I just tested them by building blocks of infill with zero perimeters, and pulling the layers apart by hand. Perimeters usually provide most of the tensile strength on small parts, but that's not always case for large parts with huge internal volumes of infill. For example, I'm making a large part that has a large flat surface on the bottom. Various points on that surface will be loaded in tension. Some of the points are far from any perimeters, so I'm mainly relying on the tensile strength of the infill. Of course I could add solid internal features near those stress points, but it would be great if the infill could take some of the stress.

unicornswag, 2020-09-29:
If I wanted to compile a custom version of SuperSlicer with Slic3r's infill styles, could I just replace all of the files in the "fill" folder with those from the current release of Slic3r?

supermerill, 2020-09-29:
No, it won't work. You can try to replace just 3Dhoneycomb.cpp but even this has very little chance to work as-is.
And you'll lost the "connect infill" bit.

unicornswag, 2020-09-29:
I've already maxed out the volumetric flow rate of my SuperVolcano hotend, so any increase in extrusion width will mean a decrease in print speed, not to mention additional wasted material. The beauty of Slic3r's improved 3D honeycomb is that it allows you to print extremely sparse, saving time and material on huge prints.

supermerill, 2020-09-29:
Erf, fill_angle_increment doesn't works, with 3D honeycomp, I have to correct also that.

An exemple with rectilinear with same direction from start to finish (but a change of infill angle for first 2mm for the exemple).
[image]

gringer, 2022-07-06:
@unicornswag you're right: cnckitchen's test used the old alternating-layer 3D Honeycomb for tests. I mentioned this in my comment in his YouTube video shortly after he published it. Unfortunately that was my buggy "beta" version of the truncated octahedron infill pattern, and it's the one that's stuck in PrusaSlicer and keeps getting forked and spread around the world.

My PR for PrusaSlicer fixes this, as well as the density calculation issue and a few other things, see here for more information:

https://github.com/prusa3d/PrusaSlicer/pull/6434

I'm happy to create a PR for SuperSlicer, but only if there's interest in it. Doesn't seem right to keep polluting bug reports with upgrade fixes when no one seems interested in them.