supermerill/SuperSlicer #507: Supporting Dense Layer - Possible Improvements

thierryzoller, 2020-09-28:
Latest Version.

Came across a use case where "Supporting Dense Layer" does (?) not work the way it should (Or rather is wasting a lot of filament to support nothing. It appears that for every small solid layer SuperSlicer adds a dense layer for the whole object, which can be quite wasteful.

See pictures and Attachement :

**Layer 81 +122 + 131 - Why a dense layer on the whole obect ?**
[image]

After layer 136 it only does dense layers.

Improvement **suggestion/idea:**
- Ideamaker lays smaller dense layers just across a certain width  : 

[image]

thierryzoller, 2020-09-28:
[supportdense.zip](https://github.com/supermerill/SuperSlicer/files/5294429/supportdense.zip)

Pavulon87, 2020-09-29:
Switch supporting dense layer algorithm to "Anchored"
[image]

thierryzoller, 2020-09-29:
Thank you, that immediately improved things =)
Would you agree however that automatic doesn't work as described and/or the description best is improved? It says it will "draw the smallest surface"- which arguably it doesn't?

supermerill, 2020-09-29:
It will create a convex shape that support the next dense infill, trying to minimize the size (abeit with a coarse optimisation, to avoid loosing too much time on it) => No turn on sparse infill.
With your object geometry, the only convex shape is the whole object, so you're better off with anchored.

That's also why the default is "auto small" where it's the automatic algorithm with a treshold to avoid filling a whole layer for a tiny support.

I'm open to suggestions if you have ideas on how to improve it.

thierryzoller, 2020-09-29:
`That's also why the default is "auto small" `

My default is Auto, not Auto small

supermerill, 2020-09-29:
I'll have to change that.

thierryzoller, 2020-09-29:
Note that Auto-Small doesn't really change much in this particular use case?

supermerill, 2020-09-29:
Not really, It's the worst case because there is nothing to anchor the denser surface unless everything is filled. And the surface to fill is significant enough (for the current threshold) to warrant a full denser infill.

Maybe the most problematic is that the speed is too low as it uses the bridge speed. It should use the bridge speed only for the perimeter and then the infill speed...

TD5023, 2020-10-21:
@supermerill I'm having issues with the dense layer trying to print on air. It does this in each mode, although it tends to miss differently. Would it be possible for the slicer to recognize where the previous layer's support lines/inner shell are so it can stretch over them and be anchored on all sides? As it stands, doing like what's in the picture doesn't help since it just droops down and leaves nothing to build on. Occasionally, it won't even attach on a single side and will extrude entirely on air.

[image]
[image]

supermerill, 2020-10-21:
in auto, it will only make strait paths between the perimeters, so it should be okay-ish.

TD5023, 2020-10-21:
Perhaps it's a bug, then, because on parts with more detailed surfaces (character statues, mostly), auto gives me the exact same issue. It typically fills the entire interior, but there will be certain instances where it seems to behave more like anchored and only attempt to lay down filament in a small spot, which is nearly always unsupported. I can get screenshots later.

TD5023, 2020-10-21:
@supermerill here are some shots of an object sliced in automatic mode. You can see where it functioned properly at a lower layer, but the circled instances show what appear to be it acting like it's in anchored mode, though with nothing to support it. If there's something else I can try, please let me know. Note that it does this for all three dense layer options.
[image]
[image]
[image]

supermerill, 2020-10-21:
can you link the project file where this happens?

TD5023, 2020-10-21:
Is the 3mf all you need for that?
[SDL example.zip](https://github.com/supermerill/SuperSlicer/files/5418803/SDL.example.zip)

supermerill, 2020-10-21:
yes

TD5023, 2020-10-21:
Cool. It's in that zip folder. If you need anything else, please let me know.

TD5023, 2020-10-30:
@supermerill 2.2.54 is still presenting issues with dense infill. The following screenshots show slicing in automatic mode, but they produce the same results with small automatic and anchored. I've also attached the project file I used. I should note that every round appearing bridge layer is behaving incorrectly. The circled one is just the one I show the alternate angle of in the second image.

[Dense layer demo.zip](https://github.com/supermerill/SuperSlicer/files/5462948/Dense.layer.demo.zip)

[image]
[image]

supermerill, 2020-11-15:
I tested your last .3mf and the issue comes from the "combine infill every 10layers", which is impossible as you can't extrude a line with a height of 2mm. I changed it to 1 and everything seems fine

TD5023, 2020-11-15:
OK, that makes sense. Since I use adaptive layer heights for everything, I intentionally use a combine infill value that will result in a larger thickness than that number of layers at the minimum height. For all the other calculations, the maximum layer height cutoff handles everything flawlessly. I've taken to just using dense infill modifiers around the problem areas. I don't intend to change the combine layer settings because I definitely don't need infill drawn out for every .04 mm layer, so I'll stay the course with modifiers. I'm assuming there isn't a reasonable way for it to detect when this would be an issue and to handle that automatically, right?

supermerill, 2020-11-15:
It should, but it don't, so it's a bug.
But I don't have the will & time to work on that in the near term.

closed 2020-11-25 by supermerill

Hello1024, 2021-01-30:
In v2.2.53, this still seems suboptimal...   Is there any reason we can't have a mode like automatic, but instead of expanding the area to a perimeter line, it simply expands to *any* line (including an infill line).

That way, the area will never be expanded beyond the infill line spacing.

For some infill patterns, there might be certain angles where it is a *long* distance to the next infill line in the worst case.   In that case, a different direction for the dense layer support should be chosen.

The algorithm *seems* simple on the surface, but I guess implementation is harder than it looks...