supermerill/SuperSlicer #510: Speed increase around more complex features

ollien, 2020-09-29:
### Version
2.2.53

### Operating system type + version
Fedora 32 - let me know if you want anything more specific here...

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro, Running Marlin 1.1.9.1

### Behavior
Printing [this model](https://www.thingiverse.com/thing:2425273) results in a really weird speed increase where the model gets more complex.
[image]

PrusaSlicer does not exhibit this behavior.
[image]

Should be able to reproduce by just slicing the model with the stock Ender 3 profile, but 
[here](https://github.com/supermerill/SuperSlicer/files/5301721/stacking-box.zip) is the 3mf/stl I used.

#### Project File (.3MF) where problem occurs
See above

Pavulon87, 2020-09-30:
It's because of "Layer time goal" in Filament Settings > Cooling

ollien, 2020-09-30:
@Pavulon87 In PrusaSlicer, I have "Slow down if layer print time is below" set to the same value as in SuperSlicer.

supermerill, 2020-09-30:
If you look, the prusaslicer has a perimeter + gapfill but there is only a thin wall on superslicer. that's because teh external perimeter width is at 0.42 in PP and 0.44 in SuSie.
Speeds are also not the same, but it's not that significant.

ollien, 2020-10-01:
Gotcha. I saw some weird print artifacts around where the speed increased, so I assumed it was the speed difference. I'll see if I can find anything else there, but it could also just be my printer.

closed 2020-10-01 by ollien