supermerill/SuperSlicer #511: std::bad_alloc on linux with "keep only bridges"

khumarahn, 2020-09-30:
Hi, I run superslicer-2.2.53 appimage.

I have a part which refuses to slice (though I successfully sliced many other things), giving me a "std::bad_alloc" error. Trying to find out what caused it, I noticed that if I change "Keep only bridges" to "Disabled" (in "No perimeters in bridge area" setting), then it slices well.

I observe the same behavior in arch linux and linux mint, but I could not reproduce the error in windows 10. To reproduce in in linux, I set up a basic profile (e.g. prusa mini) and do not change any settings except "Keep only bridges".

[cannot-slice.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5304570/cannot-slice.3mf.txt)

khumarahn, 2020-09-30:
A small update: I could reproduce it in windows, the message was "bad array new length". But when I moved the part on the table slightly, the error disappeared.

supermerill, 2020-09-30:
it's the 2.2.53.2 ?

khumarahn, 2020-09-30:
yes!

khumarahn, 2020-10-06:
I could not reproduce it with 2.53.3

closed 2020-10-22 by supermerill