supermerill/SuperSlicer #512: Add overlap setting for gapfill

supermerill, 2020-09-30:
Some report point that when a wall goes from 2 (4) perimeters to 1 (2) with a thick gap fill, the wall is too thick

neophyl, 2020-09-30:
Not just on discord.  There have been many threads on the prusa forum too.  Usually re-hashes of the same responses to new people posting about the problem.  Its far more noticeable on Prusa Slicer, your thin wall changes make it way better.

I did find this relatively recent one https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/any-tips-to-get-rid-of-the-bulge/paged/2/ 

and the grand daddy of them all I think on the prusa forum https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/buldge-when-print-reaches-solid-layers/paged/16/  

General consensus is that it happens more when you get changes between layers/perimeters that do and dont have gapfill on them iirc.

supermerill, 2020-09-30:
real print: left susie, right: pp
https://cdn.discordapp.com/attachments/461133450636951552/760858491752939550/20200930_083543.jpg

problematic part on top:

https://cdn.discordapp.com/attachments/461133450636951552/760859323898789888/unknown.png

closed 2020-10-22 by supermerill