supermerill/SuperSlicer #513: Feature Request:  Resize objects in Inches for us 'Muricans

Photogad, 2020-09-30:
Feature request for a small (possibly easy?) feature to implement. 

I sell 3D-printed items where the buyer can specify the size of the item they want. I have one single .STL file, and I load it into the slicer and re-size it there. The problem is, the slicer re-size options work in MM, which is normally fine, but my customers place their orders based on american inches, no mm or cm.

I am not so good with the conversion math in my head, so I usually have to google the inches to MM conversion and then re-size the object in slicer. It would just be a bit quicker and more convenient if there was a little toggle button I could click here to change the re-size options from MM to Inches, and vice versa. It would save me a lot of time.

How to convert millimeters to inches

1 millimeter is equal to 0.03937007874 inches:

1mm = (1/25.4)″ = 0.03937007874″

[image]

Photogad, 2020-09-30:
To be clear I'm not looking to change all of the sizes in the slicer to inches... I do PREFER to work in MM for most things. But for this, when a buyer says to me "make me this object in 7 inch size"  it would be nice to be able to just load the STL into slicer, click a button to toggle to inches for the re-size boxes, and type in the number 7. Instead of having to find out the math myself.

n8bot, 2020-09-30:
This is already in PrusaSlicer alpha, so it should be fairly easy to implement when needed.

supermerill, 2020-09-30:
If it's the next prusa, it should be merged in october.