supermerill/SuperSlicer #515: Time estimate and Filament use inconcistencies

thierryzoller, 2020-10-01:
Latest version
- Filament usage calculations wrong
- Time estimate (buffer overun? un/signed error?

[image]

thierryzoller, 2020-10-01:
This is happening repeatably after a few days of use. If I restart the error disappears. Sound like a memory managementerror ?

[usedfilament.zip](https://github.com/supermerill/SuperSlicer/files/5311633/usedfilament.zip)

supermerill, 2020-10-01:
duplicate of #364

thierryzoller, 2020-10-01:
I have good news for you, the -24xxxxxx is reproducible. Just open the file. I mistakenly did a 0.1 multiplier, wich results in these filament usage calcs. However the time should still not be zero. Can you try opening the file?

supermerill, 2020-10-01:
[image]

thierryzoller, 2020-10-01:
Interesting, would a memory dump help you find the cause?

supermerill, 2020-10-01:
not unless you put a breakpoint where the problem lie (and i don't know where it is).

ManIkWeet, 2020-10-02:
I too ran into this, curious indeed.
Maybe my file will allow you to reproduce the issue?

[negative time.zip](https://github.com/supermerill/SuperSlicer/files/5319268/negative.time.zip)

closed 2021-06-04 by supermerill