supermerill/SuperSlicer #516: Weak Support Structures

CCS86, 2020-10-01:
### Version
2.2.53

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Ultimaker Original + Duet Maestro

### Behavior
I'm struggling to get support structures that aren't doomed to fail. I am using a support enforcer block, but have it somewhat oversized. The issue seems to be that it is bounding the support so tightly to the model that it is getting clipped. This forces single line segments to stack vertically, and large gaps are created, forcing a single line width support bridge above the gap:

[image]
[image]

Is there a setting that can improve this? 

#### Project File (.3MF) where problem occurs
[35mmB.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5312079/35mmB.3mf.txt)

supermerill, 2020-10-01:
I'm not going to touch support in the near future (unless crashy bug). Not enough time for that.

n8bot, 2020-10-01:
There is this line of code which gives reference to a 1.5 mm margin, set by the macro `SUPPORT_MATERIAL_MARGIN`.

https://github.com/supermerill/SuperSlicer/blob/f3ce980fbade2eacf534cb2d684f1ffd7883bcfb/src/libslic3r/SupportMaterial.cpp#L1035-L1036

I'm not exactly sure if this is on the right path, because I also tried to play with the "expansion" of the supported areas and discovered dragons. The support patterns, before any modification, already have a tendency to "leak" out of the areas where they are supposed to be -- they will randomly fill up areas of nothingness, when one of the rectilinear lines "escapes" the region meant to be supported. Modifying this support material margin seems to exacerbate that problem immensely.

This support leak problems seems to only/mostly occur when the rectilinear angle is not 0 or 90 degrees. IIRC, modifying the margin caused it to be problematic even with 0 or 90 degrees, but I may be recalling that incorrectly.

supermerill, 2020-10-01:
So if playing with this thing, i have to do after that an `contact_polygons  = intersection(current_empty_area, contact_polygons )`  to avoid "leaks" ?

n8bot, 2020-10-01:
I have never been able to avoid the leaks, and I'm very new to c++ and coding so I would take your opinion over mine every day of the week.

All I did was change the value of the macro `SUPPORT_MATERIAL_MARGIN`. Changing that value was not helping things.

n8bot, 2020-10-01:
There is also this line (and possibly a couple other references) that use the same macro value:

https://github.com/supermerill/SuperSlicer/blob/f3ce980fbade2eacf534cb2d684f1ffd7883bcfb/src/libslic3r/SupportMaterial.cpp#L1173-L1182

supermerill, 2020-10-01:
that's the problem with the support code, it's difficult to have a good grasp on it.