supermerill/SuperSlicer #517: [Feature Request] Coasting (Reopen Feature Request #57)

robschwieb, 2020-10-01:
#57 

The above feature request is closed and marked as "fix is live in the last release" but I don't think there was a good understanding of the original request as explained by VanessaE [here.](https://github.com/supermerill/SuperSlicer/issues/57#issuecomment-684674438)

This is something that would add a lot to SuperSlicer over PrusaSlicer considering S3D, Cura, and IdeaMaker have had it for a long time. Prusa people have been requesting it for over three years [here.](https://github.com/prusa3d/PrusaSlicer/issues/246)

supermerill, 2020-10-01:
ok,ok, i'll add a slicer-based pressure model.

robschwieb, 2020-10-01:
I don't think coasting is considered a "pressure model".

I've cranked up the coasting distance to 3mm in IdeaMaker for easy visual representation but hopefully this clear it up a bit. You can see how the last 3mm of the extrusion is canceled/removed.

[image]

robschwieb, 2020-10-01:
It's not nearly as complex as that. At least not in the slicers I've used it with. Just a user specified distance (usually very small like 0.4mm) or specified volume like in Cura which recommends your nozzle diameter cubed as the value.

[image]

SnowDrifterr, 2020-10-05:
I think coasting by volume with a commanded coast speed might be most appropriate. 

I'm not sure how fruitful it would be to try to implement a pressure model given that pressure should be more or less corrected by linear advance / pressure advance. This just deals with nozzle ooze and aiding to reduce the appearance of Z seams.

With that said, I feel that a coasting calibration with this method should be relatively simple to set up. It would be as simple as printing a cube with varying coast amounts that increases by Z height, using an on-face seam alignment for reference (e.g. "back"). Or slice up a couple objects and print them individually, similar to how extrusion calibration is performed.

supermerill, 2020-10-05:
> I'm not sure how fruitful it would be to try to implement a pressure model given that pressure should be more or less corrected by linear advance / pressure advance. 

If the pressure is corrected by linear advance, then there is no need for coasting.

robschwieb, 2020-10-05:
> > I'm not sure how fruitful it would be to try to implement a pressure model given that pressure should be more or less corrected by linear advance / pressure advance.
> 
> If the pressure is corrected by linear advance, then there is no need for coasting.
> Ooze is too slow and unreliable to create the end of the extrusion.

Even with a perfectly dialed linear/pressure advance coasting still comes in handy. I get my best looking seams in S3D using just a tad of coast on top of pressure advance.

SnowDrifterr, 2020-10-06:
> Even with a perfectly dialed linear/pressure advance coasting still comes in handy. I get my best looking seams in S3D using just a tad of coast on top of pressure advance.

This has been my experience as well. I suppose I'd rather see coasting as a feature, even if basic, rather than nothing at all. That being said a well working pressure model would be sweet! 

This was posted by a peer in a 3d printer building group. Tuned pressure advance with light coasting enabled 
[image]

Itox001, 2020-10-12:
Hi, I'm the guy that opened [issue 246](https://github.com/prusa3d/PrusaSlicer/issues/246) in PrusaSlicer. As others pointed out, it is a very useful feature so if we could finally have it here it would be awesome. I have literally waited three years for this :D

I think that a simple coast as is already found on S3D or Cura would suffice. Easier to implement and it has provided a lot of value to the other slicers. Of course @supermerill if you want to do a full-blown pressure model we will happily test and report back, but I fear that the added complexity will reflect negatively on the timeline of when we could see this feature.

digitalentropy, 2020-10-15:
+1 on this for sure

boromyr, 2020-10-15:
+1 for coasting

tecnopolis-ve, 2020-11-02:
I would like to vote up for this superb feature, this one is especially useful when printing with TPU using a bowden extruder.

wildjokerss, 2020-11-12:
just came to check on this, yep, this would be wonderful. even the simplest "coasting" would be just awesome!

Zulaner, 2020-11-13:
I also want this feature

supermerill, 2020-11-13:
I have to rewrite most of the gcode generator for that, so it may take a while.

liftbag, 2020-12-23:
> I have to rewrite most of the gcode generator for that, so it may take a while.

Hopefully, I expect to be able to test it on osx when it's out, because the latest nightly builds close on opening (segmentation fault: 11)

Xploder, 2021-04-09:
coasting would be great

macchinette, 2021-04-29:
I too am voting in favor of the Coasting function. Thanks

tome9111991, 2022-04-27:
im looking for an alternative for linear advance.
because have some issue with Marlin LA extruder grinding sound.. etc..

maybe kind of coasting could be usefull to reduce pressure.
wipe not very help i have pressure by change speed for exemple by overhangs.

digitalentropy, 2022-04-27:
Re-up for coasting and whatever necessary under-the-hood changes may be needed. It can make a big difference in quality.

supermerill, 2022-04-28:
you increase the seam-gap, it's a coasting-like thing for the external periemter.

digitalentropy, 2022-04-28:
True, seam gap can compensate  slightly . It’s not clear if the seam gap value affects the beginning or end of the segment.

supermerill, 2022-04-28:
end

FDResearch, 2022-04-28:
As some feedback I have had good luck using about 220% seam gap with foaming ASA, still not perfect but this material is so weird there will always be artifacts. Honestly though on every other filament I have it off and use LA, it seems to only make gaps if your filament is properly dried.