supermerill/SuperSlicer #518: Cutting corners generate invaliad gcode with nan (not a number) in them

kursatu, 2020-10-01:
### Version
_Version of SupserSlicer used goes here_
The latest from github release, 2.2.53
_Use `About->About SuperSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
- Slice the attached project file
- Search the generated gcode for nan, you'll see things like 
G1 X276.113 Y323.126 E-nan ;  perimeter external
G1 X276.103 Y322.916 E-nan ;  perimeter external
Set the cutting corners to 0%, then the problem disappears.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for SuperSlicer, Slic3r PE 1.42.0-alpha and newer)_
[Defect11.zip](https://github.com/supermerill/SuperSlicer/files/5314196/Defect11.zip)

supermerill, 2020-10-01:
good find.

supermerill, 2020-10-01:
I also fixed a little problem witht he thin wall detection, so your print will be better now, less useless thin wall.
Also, it's better to activate overlapping perimeter with this piece, imo.

kursatu, 2020-10-06:
It seems to be fixed.

closed 2020-10-06 by kursatu