supermerill/SuperSlicer #520: Inconsistent retract and unretract when extra length parameter is set

tionebrr, 2020-10-01:
### Version
SuperSlicer_2.2.53_2020-09-18

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 running on Klipper

### Behavior
- retraction and unretraction lengths are inconsistent

Here before the first perimeter:
[image]

And after the first layers:
[image]

I have set the extra retract length as ratract+extra to get these values.
I am gladly joining the 3mf file.
#### Project File (.3MF) where problem occurs

[pot_l50.0xL50.0xH60.0_v2.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5315178/pot_l50.0xL50.0xH60.0_v2.3mf.zip)

supermerill, 2020-10-01:
there is a `G92 E0 ; reset extrusion distance` in between in the first picture, so no error.

tionebrr, 2020-10-01:
Thanks @supermerill, the problem is that my settings are as following:
[image]
(Note: I changed the extra amount from 2.05 to 2.01 since the first post, still calibrating)

tionebrr, 2020-10-01:
I am rereading my first post of this issue and it is misleading. You had to go in the parameters of the file to notice that, so my bad. I'm going to close this issue as a reference and open another more specific one.

closed 2020-10-01 by tionebrr