supermerill/SuperSlicer #521: "Extra length on restart" is applied as unretract length altogether

tionebrr, 2020-10-02:
### Version
SuperSlicer_2.2.53_2020-09-18

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 + Klipper/Mainsail

### Behavior
I am noticing that somhow, with my install, the unretract extra length is actually the unretract length.

If you set a retract length of 2mm, and add 0.01mm as extra length, the gcode shows an actual unretract length equal to the extra amount.

Shouldn't the unretract length be `retract+extra`, 2.01mm instead of 0.01mm?

[image]
[image]

supermerill, 2020-10-02:
Are you in relative or absolute E ?

tionebrr, 2020-10-02:
Aaaah thanks @supermerill, that's explaining everything.
I am in absolute.

I am using a 1mm nozzle which kind of behave strangely (underextruding). The retract/unretract was one of the first thing I checked and those values seemed weird. Didn't thought about the positioning mode.

Thanks again and sorry for the noise.

closed 2020-10-02 by tionebrr