supermerill/SuperSlicer #523: What is the purpose of Rectilinear (Solid)?

ManIkWeet, 2020-10-02:
This is more a question than a bug (or is it?).
The description in the slicer wasn't very clear.

"Rectilinear (filled) replace zig-zag patterns by a single big line & is more efficient for filling little spaces."

What I personally see it doing is adding a large amount of travel moves instead.
It seems to be extruding the lines in a single direction only. For large surfaces this seems to add a lot of printing time, especially with multiple layers doing it. I wouldn't say the print quality improved in any way.

Is it a feature that only works well on small areas? Should it be the actual default setting then?

supermerill, 2020-10-04:
rectilinear
[image]

rectilinear with gap fill
[image]

ManIkWeet, 2020-10-04:
@supermerill That makes a lot of sense!
But what I don't understand then, is that it also seems to add a lot of travel moves, is this considered a bug?

Both images have travel moves visualized
Rectilinear:
[image]

Rectilinear (filled):
[image]

supermerill, 2020-10-04:
what version do you have?
lokk at right of the infill patter, is "connected" selected?

ManIkWeet, 2020-10-04:
> what version do you have?
> look at right of the infill patter, is "connected" selected?

Version: 2.2.53
Build: SuperSlicer_2.2.53_2020-09-18

Yes, that was connected! Turning it off does remove the travel moves, so that's great!
I turned it off for the following reason:
[image]
This is a big object and the (very sparse 1%) infill was following the edge creating practically another perimeter. I didn't realise this setting also effected solid infills for top/bottom layers.

closed 2020-10-04 by ManIkWeet

supermerill, 2020-10-04:
You can modify it on a Height range modifier, if you need it.

ManIkWeet, 2020-10-04:
Ah that is very cool, I didn't know that.
I'm coming from PrusaSlicer and when messing with layer settings or object modifiers there really weren't that many settings to change. Still getting the hang of SuperSlicer! :)