supermerill/SuperSlicer #524: Wipe Tower error when multipiece parts have different number of extruders used

paukstelis, 2020-10-02:
### Version
2.2.53.2-16-g5479ce4b9

### Operating system type + version
Linux
### 3D printer brand / version + firmware version (if known)
MK3 MMU2S

### Behavior
Loading multiple part STLs where one set of parts has fewer/greater number of used extruders leads to "The Wipe Tower is only supported for multiple objects if they are sliced equally" error. Doesn't happen on PrusaSlicer.

#### Project File (.3MF) where problem occurs
[parts.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5319390/parts.3mf.zip)
The part way over to the right is the culprit.

kursatu, 2020-10-03:
I did not know that you can do this in Prusa Slicer. 
Is this a relatively new feature?
Regardless, it is something quite useful and  is apparently missing
and I've replicated the problem.
I wasn't able to open the .3mf file in Prusa Slicer, they are usable only in ps->ss direction.
So,, I've exported each individual .stl (17 of them) and then opened in
Prusa Slicer. It did not complain and sliced it fine.
SuperSlicer would not slice it, unless part4 object is made unprintable.

supermerill, 2020-10-04:
someone check float equality by using `x == y` instead of `abs(x - y) < epsilon`

closed 2020-10-22 by supermerill