supermerill/SuperSlicer #525: [Feature Request] Variable extrusion width

stefanpieter, 2020-10-03:
This would be great for:

   > small details in the print
   > better accuracy as the width can be varied as a devisiable of the part thickness.
   > automatically select thick line width for infill or other areas where resolution is not needed, which speeds up the print
   > better print quality at corners / tips, i.e. the tip is filled with line that is decreasing in width
   > better print quality  at "tunnels"/spaces between walls that dont exactly match the line width, i.e. a thicker line instead of a wiggely one

robthide37, 2020-10-03:
thin lines and gap fill are already features and can set infill bigger already do you have specific examnple of where a diff approach would be better?

supermerill, 2020-10-04:
It's useful to avoid gapfill: the gapfill can be merged with the surrounding perimeters, making them varying width perimeters.
The "better print quality at corners" is taken care in an other issue.
Other that that, I don't see something feasible of useful.

> better print quality at "tunnels"/spaces between walls that dont exactly match the line width

Use rectilinear with gap fill, it does that.

kursatu, 2020-10-05:
At first, one would think that it is similar to variable layer height.
The idea might be , if you have more curvature at the path, you decrease the line width.
@stefanpieter  One might devise a test to verify its feasibility or usefulness.
I first thought that it might be super useful. Thin external perimeters 
does improve the small feature resolution and you can print smaller features better.
But then I ran some numbers.
For a 2mm diameter feature (say a half circle), thin (0.3mm) vs. thick (0.5mm) line width, 
has a difference of only 3.5 microsteps. The extruder may not really move actually. 
The spreadsheet with calculations is attached.
One can(?) devise a test, using a mix of straight, circular paths and manually change the 
extrusion and see how it effects the small feature. It is hardly measurable though.
I have a 6200 steps/mm extruder and I can "print" 0.1 mm straight walls from the printer tests.
But, you may need more features, even from firmware. 
[Variable line width.zip](https://github.com/supermerill/SuperSlicer/files/5324902/Variable.line.width.zip)

dewi-ny-je, 2022-04-29:
This would be really useful, the advantages can be seen in Cura 5.0 and are quite impressive both for quality and speed!

robthide37, 2022-07-28:
is arachne being added ?

supermerill, 2022-07-28:
in the coming week.