supermerill/SuperSlicer #529: Wrong geometry in some layers

Kurolox, 2020-10-05:
### Version
Version 2.2.53

### Operating system type + versionSuperSlicer
Version:   2.2.53
Build:     SuperSlicer_2.2.53_2020-09-18

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 20226), 64-bit edition
Total RAM size [MB]: 11,753MB
OpenGL installation
GL version:   4.6.14736 Compatibility Profile Context 20.9.2 27.20.12031.1000
Vendor:       ATI Technologies Inc.
Renderer:     Radeon RX 590 Series
GLSL version: 4.60
Installed extensions:
GL_AMDX_debug_output
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_gcn_shader
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_interleaved_elements
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_occlusion_query_event
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_atomic_counter_ops
GL_AMD_shader_stencil_export
GL_AMD_shader_stencil_value_export
GL_AMD_shader_trace
GL_AMD_shader_trinary_minmax
GL_AMD_sparse_texture
GL_AMD_sparse_texture_pool
GL_AMD_stencil_operation_extended
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_transform_feedback4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
[... 246 lines cut ...]
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Modified Prusa i3, running Marlin bugfix-2.0.x (Sep 23 2020 20:39:02)

### Behavior
Some layers seem to be generating wrong for no apparent reason. While the screenshot from the preview has been generated at .02mm layer height for ease of displaying, this happens too in higher layer heights. I'm using the default superslicer profile with no adjustment made (besides the layer height, as stated previously)

Here's the model, previous to slicing:
[image]

Here it can be seen that the ear has some weirdly generated layers that do not match the model:
[image]

Here's another part of the model with a layer with issues: 
[image]

It doesn't seem to be an issue with the .stl file (no errors were detected), but I've provided it below just in case.

#### Project File (.3MF) where problem occurs
[3mf.zip](https://github.com/supermerill/SuperSlicer/files/5328271/3mf.zip)
.STL of the shown model: [mini.zip](https://github.com/supermerill/SuperSlicer/files/5328326/mini.zip)

supermerill, 2020-10-05:
the model has empty space between the ear and the head, i think that's the problem.
Please select a single slice (0.02mm), set the first layer height/width to the normal oneand, check if the slice is wrong or not.
If wrong, please put the on-slice .3mf here.

Kurolox, 2020-10-05:
It seems like it's wrong indeed.

[image]

[image]

Here's the .3mf with the slice: [slice.zip](https://github.com/supermerill/SuperSlicer/files/5329125/slice.zip)

supermerill, 2020-10-05:
[image]
red: perimeter middle line
yellow: the object shape
cyan: area without perimeter extrusion
blue : area where there is enough width for a thin_wall

closed 2020-10-08 by Kurolox