supermerill/SuperSlicer #532: gcc version requirements not set in cmake files

TheAndr0id, 2020-10-06:
### Version
2.2.54.0

### Operating system type + version
Linux x86_64 custom (was Slackware 14.2 x64)

### Behavior
Currently the cmake files don't check to see what version of gcc/g++ is being used.  Some of the code is using the gnu++1z standard which has been a moving target for the last few years.  I was using gcc 5.5 (old, but most things don't care), and libslic3r failed to build with the following message:

```
SuperSlicer-2.2.54.0/src/libslic3r/MeshBoolean.cpp: In function 'void Slic3r::MeshBoolean::minus(Slic3r::MeshBoolean::EigenMesh&, const EigenMesh&)':
SuperSlicer-2.2.54.0/src/libslic3r/MeshBoolean.cpp:60:11: error: expected unqualified-id before '[' token
     auto &[VA, FA] = A;
```
The "auto &[ ... ]" expression is a newer addition to the gnu++1z standard.  When I updated to a newer gcc, the code complied.  This should be included as a requirement in the cmake files.

I'm sorry, I don't know where/when this was added to gcc to give you a minimum version requirement.

Once again, this might need to be up-streamed into PrusaSlicer and Slic3r.

supermerill, 2021-05-20:
how to add it? where?

TheAndr0id, 2021-05-21:
I'm not a cmake guy (don't get me started...  still much better than the perl method Slic3r is using).  I did a quick google and found this link:

[https://stackoverflow.com/questions/14933172/how-can-i-add-a-minimum-compiler-version-requisite](url)

The last comment stated this:

> You can check the specific gcc version as follows:
> 
> ```
> if (CMAKE_COMPILER_IS_GNUCC AND CMAKE_CXX_COMPILER_VERSION VERSION_LESS 5.1)
>     message(FATAL_ERROR "Require at least gcc-5.1")
> endif()

I did a quick poke on the cmake.org documentation and these directives seem to be there.  The comments from the above link were in 2013/2017...

Another idea would be to put in some checks for the specific standard you are using.  Now in the case above I found that the standard was in the older version of GCC, it just wasn't the *same* standard as the current...

The cmakelists.txt file has a line: 
`cmake_minimum_required(VERSION 3.8)`

that might be a good place to put it?  (I'm the wrong guy to ask, I know VERY little cmake)

supermerill, 2021-05-21:
I mean, if you can test it on your computer that it correctly output a fatal error message.
I can't add a line that I don't know if it works.

TheAndr0id, 2021-05-21:
So I've upgraded my gcc, and although it's introduced different issues (excessive warnings), rolling it back is something that I'm not all that keen on.

TheAndr0id, 2021-05-24:
I just took a look at the current CMakeLists.txt, and there is now a check for version of GCC in it.  Around line 190ish there's the following:

>     # removes LOTS of extraneous Eigen warnings (GCC only supports it since 6.1)
>     # https://eigen.tuxfamily.org/bz/show_bug.cgi?id=1221
>     if("${CMAKE_CXX_COMPILER_ID}" MATCHES "Clang" OR CMAKE_CXX_COMPILER_VERSION VERSION_GREATER 6.0)
>         add_compile_options(-Wno-ignored-attributes) # Tamas: Eigen include dirs are marked as SYSTEM
>     endif()

When I entered this issue I was using gcc 5.5, and this check is obviously new (requiring gcc 6.0).  This looks to have been pulled in from the Prusa source, so this issue might have already been dealt with by them.

That being said, the CMakeLists.txt from Prusa seems to be newer than yours.  That may mean nothing, but it might be worth taking a look to see if they've added any additional important requirements for their 2.3.1 build.

In any case I think this issue can be closed.  I'll let you confirm that...

closed 2021-05-25 by supermerill