supermerill/SuperSlicer #534: colorprint doesn't change default color

wnc2, 2020-10-06:
### Version
2,54

### Operating system type + version
MacOS catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
MK3S with palette

### Behavior

it is impossible to change the default extruder in coloprint, if i change, the right color blinks for a second, then returns grey again, no problem for additional layer change.
Feature is ok in the last 2.53 and previous

[image]

_Is this a new feature request?_

no

supermerill, 2020-10-06:
yeah, i think it's the same bug as the other one. I initialized the extruder too early...
I'm correcting it.

wnc2, 2020-10-06:
Hope this helps

Bye

⁣Ottieni BlueMail per Android ​

closed 2020-10-28 by supermerill