supermerill/SuperSlicer #536: Cold extrusion on print

mattchan-tencent, 2020-10-07:
### Version
2.2.54

### Operating system type + version
MacOS 10.15.6

### 3D printer brand / version + firmware version (if known)
Creality Ender 5 Pro + Marlin 2.0.7
- direct drive
- OEM extruder
- linear rails
- skr v1.4 turbo mainboard

### Behavior
Generated gcode is missing M109/104 instruction. Prints are extruded cold.

Steps to reproduce:
- create custom printer with marlin firmware (no preset for ender 5 included)
- set filament temperature (e.g. 210)
- slice stl
- export gcode. M109/104 will be missing. M190 (bed heating) is there though.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[pins.zip](https://github.com/supermerill/SuperSlicer/files/5337897/pins.zip)

neophyl, 2020-10-07:
Try adding the following into your start gcode in printer settings and removing your manual M109 that you have in there.

M140 S[first_layer_bed_temperature] ; set bed temp
M104 S[first_layer_temperature] ; Set Extruder Temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

Those are the place holders for the temperature settings as configured in your chosen filament.

mattchan-tencent, 2020-10-07:
I understand I can add it myself, but shouldn't superslicer add that gcode if it's missing? For example if I change the filament temperature in the GUI, it won't have any effect on the gcode right?

My manual M109 was added after the failed print. If I take it out, it cold extrudes.

marekorok, 2020-10-16:
Hi,
I have the same issue, checked the generated gcode and seems there is M104 (no wait) while should be M109 (wait)
Isn't a bug?
```
; generated by SuperSlicer 2.2.53 on 2020-10-16 at 07:49:18 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.45mm

M73 P0 R11
M107 ; disable fan
M190 S70 ; set bed temperature and wait for it to be reached
M104 S245 ; set temperature
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0 ; reset extrusion distance
;LAYER:0
M117 INDICATOR-Layer0
```

supermerill, 2020-10-16:
@marekorok i need your project file

marekorok, 2020-10-16:
> @marekorok i need your project file
sure
[test.zip](https://github.com/supermerill/SuperSlicer/files/5391094/test.zip)

fyi, the only custom g-code I have is the 'Before layer change"

neophyl, 2020-10-16:
I think that's your problem.  You have absolutely no g-code in the Printer Start or End sections.  You need to add in suitable g-code for your printer.  Stuff to turn the extruder and bed heaters off in the end gcode for example, the filament heat up lines posted previously into the start up area etc.

Also looks like something in the setup/file is also messing up the remaining time value (the M73 P0 R section between the 2 versions as its a large negative value in the 2.2.54 version ).

marekorok, 2020-10-16:
I could agree, but... I never had any extra gcode here 
and it was working fine till 2.2.53, then something happened on 2.2.54.
I've checked also PrusaSlicer, also generates 2 commands, as 2.2.53
```
; generated by PrusaSlicer 2.2.0+win64 on 2020-10-16 at 15:59:31 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.70mm

M73 P0 R24
M73 Q0 S23
M107
M190 S70 ; set bed temperature and wait for it to be reached
M104 S245 ; set temperature
M109 S245 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
```

Not a big deal, if you don't waste time here you can skip it.
I can add the code.
Still the best slicer ever! :-)

supermerill, 2020-10-16:
@marekorok same bug as the other reports, already fixed in the nightly & same bug as written in the release page.

closed 2020-10-16 by supermerill