supermerill/SuperSlicer #538: Retraction Calibration not correctly generating.

afkfurby, 2020-10-07:
### Version
2.2.54 

### Operating system type + version
win10

### Behavior
The retraction calibration is generating some strange plate below the actual test and has an air gap.
That way you get an empty layer detected while slicing

[image]

afkfurby, 2020-10-07:
ah found out what this is, It's the temperature label placed at -1.33 instead of 0 or +0.4

supermerill, 2020-10-07:
can you post your project file ? i can't reproduce... it's with a big or little nozzle diameter?

zervin, 2020-10-21:
Seeing the same issue with a 0.5mm nozzle.
[retraction_calibration.zip](https://github.com/supermerill/SuperSlicer/files/5417290/retraction_calibration.zip)

closed 2020-10-28 by supermerill