supermerill/SuperSlicer #540: Ejecting an SD card is slow and locks UI

BeansIsFat, 2020-10-07:
### Version
2.2.53

### Operating system type + version
Mac OS 10.14.6

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
Ejecting an SD card takes a long time and locks the UI.

Steps to reproduce:
* Insert SD card
* Export G-code to SD card
* Click Remove Device button

Expected result:
* Quick eject
* Responsive UI

Actual result:
* Slow eject
* UI non-responsive
* Old-style watch cursor
* Spinning beachball cursor

#### Project File (.3MF) where problem occurs
N/A

It does not appear to be an issue with my computer as ejecting via Finder is almost instantaneous.

supermerill, 2020-10-07:
Thank you for the report.
Not going to lie, but this won't be a priority for me, hoping that prusa will improve it before I have time to look at it.

BeansIsFat, 2020-10-07:
That's fair. If this eject code is identical in PrusaSlicer then I'll report it over there as well. Thanks.

supermerill, 2020-10-07:
I took a quick look, in mac all it does is calling `diskutil eject /path/to/gcode/file`
can you test in a terminal if this command is really much quicker than the gui?

BeansIsFat, 2020-10-07:
That command works at about the same speed as Finder and much quicker than the GUI.

BeansIsFat, 2021-02-11:
Fixed by 18cf1fdb43cd6940733fdda8a4c1b07adff23b0e

closed 2021-02-11 by BeansIsFat