supermerill/SuperSlicer #542: Bug: First layer temperature only applying for Skirt - and not to layer 1

Saqib117, 2020-10-07:
### Version
2.2.54

### Operating system type + version
Windows 10 64-bit

### 3D printer brand / version + firmware version (if known)
Voron 2.4

### Behavior - Bug
Layer 1 temperature no longer being applied to first layer of the print model, only to the skirt - confirmed this works correctly in 2.2.53, and changing no other variables the slicer preview and exported gcode changes to print temperature as soon as the model print begins and not on layer 2 as expected.

supermerill, 2020-10-07:
most problably the same bug as #533 and #534. Try the nightly build ("Actions" menu)

digitalentropy, 2020-10-08:
The bug still appears to be present on the latest nightly, although #533 seems to have been resolved by the latest commit.

closed 2020-10-28 by supermerill