supermerill/SuperSlicer #543: Idea: Volumetric flow calibration

Czarnyszarf, 2020-10-08:
As title says - Maximum volumetric flow is important setting, responsible for maximum allowed speed on the printer and it varies from filament to filament.

It could be done by printing a rather large square requiring around 10-15cm of filament  with higher flow each layer. 
The layer change could be announced by printers speaker (ex. 5th layer - five beeps) or by pause on each layer

How to recognize correct settings: If user hears extruder cracking/sees under extrusion or filament not feeding consistently then the flow is too high. 

The square itself could be made from high layer to avoid super-fast movements ( calculator: https://www.printitindustries.com/pages/print-speed-calculator)

supermerill, 2020-10-08:
imo, problems occur before the extruder click. The filament isn't fluid enough and the corners began to be strange and the layer adhesion began to be bad. And I don't think there is a threshold but more of a continuous area where the problems worsen.

Czarnyszarf, 2020-10-08:
When I calibrate I look at the pauses when filament is feeding , that means that it's too much.

supermerill, 2020-10-08:
the pauses?

Czarnyszarf, 2020-10-08:
As you look into the filament that goes into the extruder - when you see its not going in consistently , has small "pauses" when feeding. It means there is volumetric blockage in hotend. 

You can test it yourself, give printer order like G1 e150 F300 and look at the extruder, then go lower, f280 etc.

supermerill, 2020-10-08:
can you give a gcode exemple (ideally, for an e3D nozzle)?

Czarnyszarf, 2020-10-08:
I did it manually most of the time.
first, heat up your hot end, raise it a decent bit above the bed, then use your host software to repeatedly extrude 50+mm, slightly increasing the speed each time. repeat until the extruder motor starts skipping steps, or the filament starts grinding

gcode: G1 E150 F300 , f280 etc. 
let's say our max speed before the extruder started skipping was 3mm/s (note: this is in terms of mm of filament going in to the extruder, not out). To calculate the volumetric flow, just multiply your max speed by the area of your filament.

max input feedrate * pi * (filament diameter/2)2 = max flowrate

I think you can make it more automated with printing square winch has different flow rate on each layer. 

in my case, it would be: 3 * 3.14 * 0.8752 = 7.21 mm/s3

ProfEngr, 2020-10-08:
@Czarnyszarf Were you following this video by Teaching Tech during your tests? [https://youtu.be/Mnvj6xCzikM?t=146](url)

Czarnyszarf, 2020-10-08:
@ProfEngr  no, but thats very useful! That excel sheet will come handy next time.

ProfEngr, 2020-10-08:
@Czarnyszarf I'd forgotten that I had to correct a typo in one of the formulas in that sheet. Maybe he's fixed his by now.

ERamseth, 2022-08-09:
Sorry to necro this but I guess it's better than making a new issue.

I think something like what KISSlicer does would work GREAT. I actually still have a coipy of KISSlicer just to do calibration of flow, temperature, retractions, etc. 

Basically it has the functionality of adjusting whatever parameter you want to tune based on height where you set a range within which you want to calibrate (for example, 1-10 mm3/s for flow) and it adjusts that variable based on height when slicing. So the bottom layer might be 1mm3/s and the top might be 10mm3/s and you just print any object (cube with rounded corners is suggested) and observe the height at which the print quality goes haywire, come down a little from that, and then you have a good value for whatever parameter you want to tune.  Calipers highly recommended, but you could probably eyeball it with a good ruler.

Relevant links

https://www.kisslicer.com/

https://snolabs.com/blog/the-missing-manual-to-kisslicer/
(scroll to bottom or ctrl+F for tuning wizards)