supermerill/SuperSlicer #545: Change in infill percent doesn't ask to reslice

dkatzdev, 2020-10-08:
### Version
2.2.54

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
After slicing a model, changing the infill percentage does not trigger a gcode reset which would normally ask for you to reslice the model. For example, I loaded the lack enclosure project file, changed infill from Grid to Gyroid. This requires reslicing. Then when changing infill from 10% to 20% it does not ask me to reslice. Changing back from 20% to 10% still does not ask for a reslice.

#### Project File (.3MF) where problem occurs

[prusa_mmu_enclosure_10.zip](https://github.com/supermerill/SuperSlicer/files/5348573/prusa_mmu_enclosure_10.zip)

supermerill, 2020-10-16:
On the latest nithgly
I loaded your .3mf, i changed top & bottom to monotnous, i sliced, i changed the infill %, and the infill step get reinitialized.

what I missed?

supermerill, 2021-06-26:
can't reproduce, please reopen if still here.

closed 2021-06-26 by supermerill