supermerill/SuperSlicer #546: Supporting Dense Layer too aggressive.

LaserBee, 2020-10-08:
Latest version (happened in previous ones too, I think...

[image]

Infill "supporting dense layer" is way too keen. It covers the whole interior every few layers (between five and 15), sometimes even with five or six layers of "Bridge infill". It only really needs to do this towards the edge, and surely not so many layers at a time!

LaserBee, 2020-10-08:
[image]
[image]

liftbag, 2020-10-08:
Have you tried selecting "Anchored" rather than "Automatic"?

Automatic

[image]

Anchored

[image]

supermerill, 2022-03-23:
there is now more options and the algo is refined.

closed 2022-04-01 by supermerill